pyusb/pyusb #199: Feature: usb.control.get_status() does not have any option to get non-standard descriptors

Jude188, 2018-05-04:
I am working with USB Hubs and am wanting to get usb hub's 'hub descriptor' as specified in the USB spec. I cannot just use:
```python
usb.control.get_status(hub, 1024, usb.DT_HUB, 0)
```
because the bmRequestType is hard coded to be generated with:
```python
bmRequestType = usb.util.build_request_type(
                    util.CTRL_IN,
                    util.CTRL_TYPE_STANDARD,
                    util.CTRL_RECIPIENT_DEVICE)
```
The util.CTRL_TYPE_STANDARD is the problem here. I need to use util.CTRL_TYPE_CLASS. This seems like a feature which could be fairly easily added and I am happy to look at this myself if you think that it would be a good addition. However, I thought it would be good to start a discussion about how to deal with the issue before starting. Should there be an additional argument to determine which CTRL_TYPE to use?   

For the moment my solution is to generate the bmRequestType myself and use
```python
bmRequestType = usb.util.build_request_type(
                    usb.util.CTRL_IN,
                    usb.util.CTRL_TYPE_CLASS,
                    usb.util.CTRL_RECIPIENT_DEVICE)

wValue = usb.DT_HUB << 8

dev.ctrl_transfer(bmRequestType, usb.REQ_GET_DESCRIPTOR, wValue=wValue, wIndex=0, data_or_wLength=1024)
```

I'm sure that this issue probably spans across at least the rest of the descriptor functions in usb.control and possibly others too.

Any feedback or comments would be welcomed! I've really only just started using pyusb!

mcuee, 2019-12-07:
Seems to be a valid request and we can add a additional entry in the pyusb/control.py file.

Jude188, 2019-12-07:
When you say an additional entry do you mean an additional argument? Because looking at this (https://www.beyondlogic.org/usbnutshell/usb6.shtml) it seems like really CTRL_TYPE_STANDARD should not be hardcoded anywhere. If this added as a keyword argument with a default value of CTRL_TYPE_STANDARD it should't result in a breaking change. 

Though that is a little ugly. Doesn't seem worth introducing a breaking change though :/

Anyway, this is just my thoughts having not looked or thought about this for about a year and a half!

Jude188, 2019-12-07:
Please do let me know if you agree that this issue is more widespread than just descriptors in the control functions.

jonasmalacofilho, 2019-12-08:
@jude188

> Because looking at this (https://www.beyondlogic.org/usbnutshell/usb6.shtml) it seems like really CTRL_TYPE_STANDARD should not be hardcoded anywhere. `get_status`, `get/set_configuration`, `get/set_interface`, or `clear/set_feature` can all take Class or Vendor options, I believe.
> [...]
> Please do let me know if you agree that this issue is more widespread than just descriptors in the control functions.

Actually I don't agree, as these methods map to _standard device requests._

While the spec states:

> The USB Specification defines a series of standard requests that all devices must support. These are enumerated in Table 9-3. In addition, a device class may define additional requests. A device vendor may also define requests supported by the device.

, these additional requests may _or may not_ match the names and descriptions of their standard counterparts.

jonasmalacofilho, 2019-12-08:
Class-specific requests accepted by devices from the Hub class:

 - CLEAR_FEATURE: ClearHubFeature (recipient=Device); ClearPortFeature (recipient=Other)
 - GET_DESCRIPTOR: GetHubDescriptor
 - GET_ STATUS: GetHubStatus (recipient=Device); GetPortStatus (recipient=Other)
 - SET_DESCRIPTOR: SetHubDescriptor
 - SET_ FEATURE: SetHubFeature (recipient=Device); SetPortFeature (recipient=Other)

The base USB 2.0 spec doesn't specify class-specific requests for other classes, though I'm sure other documents do.

Still, I think that as our methods map standard requests, they should at most implement standard + hub requests (also only `get/set_descriptor`, `get_status` and `clear/set_feature` should be affected).

jonasmalacofilho, 2019-12-08:
Note: the latest USB 3.2 spec has a different list of class-specific requests for hubs, lacking the transaction translator (TT) requests but also adding GetPortErrorCount and SetHubDepth.  My first thought is can simply support the union of these, but maybe that's not safe and the hub type has to be checked/validated.

mcuee, 2019-12-09:
I think we can implement only standard requests. The others are optional.

mcuee, 2021-07-24:
@jonasmalacofilho Just wondering if you can take a look at this again. Thanks.

jonasmalacofilho, 2021-07-24:
I think my position is still the same: we can implement the hub requests, but we should probably stop there.  Other/custom requests should generally be implemented by the user; or, perhaps in the future, with methods specific for device classes of interest.

(The hub requests may share APIs with standard device requests in cases like `SET_FEATURE`).

jonasmalacofilho, 2021-07-24:
@judahrand is this still something you're interesting in doing?

lucasdimitrov19-lang, 2026-10-24:
# Ein Repository forken

Erstellen Sie auf GitHub einen Fork eines Repositorys, um Änderungen vorzuschlagen, an Projekten zusammenzuarbeiten und Ihre eigene Kopie der Codebasis zu verwalten.

## Informationen zu Forks

Wenn Sie ein Repository forken, können Sie Änderungen an einem Projekt vorschlagen, ohne dass sich dies auf das Upstream-Repository auswirkt. Siehe [Informationen zu Forks](/de/pull-requests/get-started/about-forks).

## Voraussetzungen

Falls Sie dies noch nicht getan haben, richten Sie Git und die Authentifizierung mithilfe von GitHub.com von Git ein. Siehe [Einrichten von Git](/de/get-started/git-basics/set-up-git).

## Ein Repository abzweigen

<div class="ghd-tool webui">

Man erstellt vielleicht einen Fork eines Projekts, um Änderungen am Upstream-Repository vorzuschlagen. In diesem Fall empfiehlt es sich, Ihren Fork regelmäßig mit dem Upstream-Repository zu synchronisieren. Zu diesem Zweck musst du Git in der Befehlszeile verwenden. Du kannst das Festlegen des Upstream-Repositorys mit demselben [octocat/Spoon-Knife](https://github.com/octocat/Spoon-Knife)-Repository üben, das du gerade geforkt hast.

1. Gehen Sie auf GitHub zum [octocat/Spoon-Knife](https://github.com/octocat/Spoon-Knife)-Repository.
2. Klicke in der oberen rechten Ecke der Seite auf **Fork**.

   [image]
3. Wähle unter „Besitzer“ das Dropdownmenü und dann eine*n Besitzer*in für das geforkte Repository aus.
4. Standardmäßig erhalten Forks den gleichen Namen wie die zugehörigen Upstream-Repositorys. Geben Sie optional im Feld "Repository-Name" einen anderen Namen ein, um Ihren Fork zu unterscheiden.
5. Gib optional im Feld „Beschreibung“ eine Beschreibung für deinen Fork ein.
6. Wähle optional **Nur Standardbranch kopieren** aus.

   Bei vielen Fork-Szenarien, wie z. B. bei Beiträgen zu Open-Source-Projekten, genügt es oft, den Standardbranch zu kopieren. Wenn du diese Option nicht auswählst, werden alle Branches in den neuen Fork kopiert.
7. Klicke auf **Fork erstellen**.

> [!NOTE]
> Du kannst über die Seite **Branches** weitere Branches aus dem Upstreamrepository kopieren. Siehe [Verwalten von Filialen innerhalb Ihres Repositorys](/de/pull-requests/how-tos/commit-changes/managing-branches-within-your-repository).

</div>

<div class="ghd-tool cli">

> [!NOTE]
> Weitere Informationen zur GitHub CLI findest du unter [Informationen zu GitHub CLI](/de/github-cli/github-cli/about-github-cli).

Um einen Fork für ein Repository zu erstellen, verwende den Unterbefehl `gh repo fork`.

```shell
gh repo fork REPOSITORY
```

Um den Fork in einer Organisation zu erstellen, verwende das `--org`-Flag.

```shell
gh repo fork REPOSITORY --org "octo-org"
```

</div>

<div class="ghd-tool desktop">

Sie können auf GitHub.com oder in GitHub Desktop einen Fork eines Repositorys erstellen. Informationen zum Forken auf GitHub.com finden Sie in [der Webbrowser-Version dieses Artikels](/de/pull-requests/how-tos/work-with-forks/fork-a-repo?tool=webui).

Wenn du in GitHub Desktop ein Repository klonst, für das du nicht über Schreibzugriff verfügst, und dann versuchst, eine Änderung an das Repository zu pushen, wird ein Fork für dich erstellt.

1. Klicke im Menü **File** auf **Repository klonen**.

   <div class="ghd-tool mac">

   [image]

   </div>

   <div class="ghd-tool windows">

[... 177 lines cut ...]
* In dein persönliches Konto
* In eine Organisation, in der du über die Berechtigung zum Erstellen von Repositorys verfügst

Wenn du Zugriff auf ein privates Repository hast und der Besitzer das Forken zulässt, kannst du das Repository forken:

* In dein persönliches Konto
* An eine Organisation auf GitHub Team, bei der Sie Berechtigungen zum Erstellen von Repositorys haben

Sie können mithilfe von GitHub Free kein privates Repository zu einer Organisation forken. Weitere Informationen zu GitHub Team und GitHub Free, finden Sie unter [Pläne von GitHub](/de/get-started/learning-about-github/githubs-plans).

Weitere Informationen dazu, wann Sie ein Repository verzweigen können, finden Sie unter [Gabeln](/de/pull-requests/reference/forks).

Sie können unter [Entdecken GitHub](https://github.com/explore) nach Projekten suchen und anfangen, zu Open-Source-Repositories beizutragen. Siehe [Möglichkeiten finden, Open Source auf GitHub beizutragen](/de/get-started/exploring-projects-on-github/finding-ways-to-contribute-to-open-source-on-github).

## Nächste Schritte

Du hast jetzt ein Repository geforkt, das Klonen deines Forks geübt und ein vorgelagertes Repository konfiguriert.

* Weitere Informationen zur Verwendung von Git in der Befehlszeile zum Klonen und Synchronisieren von Änderungen finden Sie unter [Einrichten von Git](/de/get-started/git-basics/set-up-git).

* Sie können auch ein neues Repository erstellen, um Ihre Projekte zu speichern und den Code auf GitHub zu teilen. Durch das Erstellen eines Repositorys für dein Projekt kannst du Code auf GitHub speichern. Auf diese Weise wird die Arbeit abgesichert, die du mit anderen Entwickler\*innen teilen kannst. Weitere Informationen finden Sie unter [Schnellstart für Repositorys](/de/repositories/creating-and-managing-repositories/quickstart-for-repositories).

* Jedes Repository auf GitHub gehört einer Person oder einer Organisation. Sie können mit Personen, Repositories und Organisationen interagieren, indem Sie sich auf GitHub mit ihnen vernetzen und ihnen folgen. Weitere Informationen finden Sie unter [Entdecken von Projekten auf GitHub](/de/get-started/exploring-projects-on-github/discovering-projects-on-github).

* GitHub verfügt über eine großartige Supportcommunity, in der du um Hilfe bitten und mit Menschen aus der ganzen Welt sprechen kannst. Nimm an der Unterhaltung auf [GitHub Community](https://github.com/orgs/community/discussions) teil.