pyusb/pyusb #200: How to use usbdk as backend on Windows 10

abulka, 2018-05-23:
PyUsb needs a 'backend' installation step which is simple on Mac (`brew install libusb`) but is not well documented for windows. It seems the leading candidates are libusb-win32 vs [libusb-1.0.22](https://libusb.info/), the latter seems to be the latest/official one - right? which has a [recommended installer, zadig](https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows).

Unfortunately zadig replaces the default windows Usbprint.sys driver, which I don't want to do.  So am trying to follow the advice #186 

> For libusb 1.0 Windows backend, you can use usbdk backend to keep the existing driver.

So I installed usbdk from https://github.com/daynix/UsbDk/releases and running `C:\Program Files\UsbDk Runtime Library\UsbDkController.exe -n` successfully lists my device.  So far so good.

However it seems that I still need a libusb-1.0 Windows library https://sourceforge.net/p/pyusb/mailman/message/36242674/ 

> libusb-1.0.dll is the user space library on top of the
> usbdk/winusb/libusbk for you to easy access the device.

So, bypassing the zadig installer, I downloaded the dlls from https://libusb.info/ and copied all the `...\libusb-1.0.22\MS64\dll` files into my Python project and ran a short Python script

    import usb.core
    import usb.util
    import usb.backend.libusb1
    backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
    print(backend)
  
    # find our device
    dev = usb.core.find(idVendor=0x04b8, idProduct=0x0e03, backend=backend)

but this gives me 'NotImplementedError: Operation not supported or unimplemented on this platform' - meaning I think that something is not connected.  

**Not** having the libusb-1.0.dll in my Python directory and running the script causes the famous no backend found error, so at least I'm not getting that!  According to the [usbdk docs](https://www.spice-space.org/download/windows/usbdk/UsbDk_Software_Development_Manual.pdf) clients programs should access usbdk via `UsbDkHelper.dll` but

    backend = usb.backend.libusb1.get_backend(find_library=lambda x: "C:\\Windows\\System32\\UsbDkHelper.dll")

didn't work either?  

Python 3.6 (64bit), Windows 10 (1803) (64bit), trying to talk to an Epson TM-TS20 thermal printer connected via USB.

abulka, 2018-05-24:
I've been [advised](https://github.com/pbatard/libwdi/issues/127) that in order to use the usbdk driver, libusb must be put into usbdk 'mode' via a call to `libusb_set_option` immediately after `libusb_init`. However pyusb 1.0.2 does not seem to expose `libusb_set_option` - what am I to do?  Are there plans to expose this endpoint?

abulka, 2018-05-29:
I managed to activate usbdk mode:

        import usb.backend.libusb1
        from ctypes import c_void_p, c_int
        backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
        backend.lib.libusb_set_option.argtypes = [c_void_p, c_int]
        backend.lib.libusb_set_option(backend.ctx, 1)  # <--- this is the magic call to enable usbdk mode
        self.device = usb.core.find(idVendor=self.idVendor, idProduct=self.idProduct, backend=backend)
        log.debug(self.device)  # should show lots of nice device info

Ideally pyusb would be updated to add the one line which configures the argtypes for the `libusb_set_option` call rather than my having to 'patch it' in as the above code does.

The actual call to `libusb_set_option` which enables the use of the nice, non impacting [usbdk driver](https://github.com/daynix/UsbDk/releases) needs `1` as the second parameter which corresponds to the enum `LIBUSB_OPTION_USE_USBDK` - [see](https://github.com/libusb/libusb/blob/fef88bf2b91ee4d7cfcc9d3b799a2996ecfe0d2c/libusb/libusb.h).

Thus:

    python --> pyusb (pure python) --> libusb-1.0.dll (C API) --> usbdk (driver)

mcuee, 2018-06-11:
Ref: http://libusb.sourceforge.net/api-1.0/group__libusb__lib.html#ga07d4ec54cf575d672ba94c72b3c0de7c

So it is not really a magic. But I agree the documentation of libusb can be improved.

```
enum libusb_option (LIBUSB_OPTION_LOG_LEVEL = 0; LIBUSB_OPTION_USE_USBDK = 1)

mcuee, 2018-08-01:
Wiki updated.
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

mcuee, 2018-08-07:
Probably pyusb can be updated to allow for the use now that libusb has the changes in place.

RyanHope, 2019-07-26:
Has anyone else been able to get usbdk working? I get an error when I call libusb_set_option:
Procedure probably called with too many arguments

mcuee, 2019-07-30:
You can try to patch code of pyusb using what abulka mentioned in previous comments.

RyanHope, 2019-09-27:
I finally got this working. I guess with usbdk you can not mix 32bit and 64bit. With zadig/libusbk I could use 32bit python on windows and 32bit libusb and I could build and app that worked fine on win32 or win64. With usbdk if windows is 64bit I have to use 64bit python.

RyanHope, 2019-09-27:
Correction, this just does not seem to work in win32 at all.

RyanHope, 2019-09-27:
Finally figured out a solution for win32. The va_list (...) aspect of the libusb_set_option is not really supported by ctypes. Not sure why this works on 64bit but 32bit definitely doesn't like it. So, what I did what add a libusb_set_usbdk(*ctx) function to libusb which has a simpler function signature. After building and linking to this in my app I can run a 32bit python with 32bit libusb on win10 64bit with 64bit usbdk.... Yay!

mcuee, 2019-11-29:
In that case, you may want to submit a patch to libusb to see if that can be accepted.

RyanHope, 2019-11-29:
I considered that but since its not a libusb bug I didnt think they would care. The bug is in ctypes.

mcuee, 2019-12-07:
I see. Probably you can feedback to ctypes if that is indeed a ctype bug. Or maybe there is a workaround for 32bit.

jonoboy9, 2020-05-21:
I am trying to get PYusb libusb1 to be in libusbk DLL on Windows 10 using Python 3.6 and 3.8.
I installed libusbk using Zadig. Having connect my Nikon camera, in Zadig, I listed all devices, selected the Nikon driver and replaced it by libusbk. Is what I have done correctly.
The version of libusbK is 3.0.7.0 and the version of the libusb-1.0.dll is  1.0.22.11312.
Pyusb is 1.0.2 as in pyusb-1.0.2-py3.8.egg-info

I am running the suggested script:
import usb.backend.libusb1
from ctypes import c_void_p, c_int
backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
backend.lib.libusb_set_option.argtypes = [c_void_p, c_int]
backend.lib.libusb_set_option(backend.ctx, 1) 

On attempting to set the option, it returns -5 which I believe is LIBUSB_ERROR_NOT_FOUND.
Everything is 64 bit.
Just to double check, I used dllexp to dislay all exported functions from libusb-1.0.dll and the libusb_set_option was present.
I gather this must have been working. What have I done wrong? I do appreciate help. Many thanks

jonasmalacofilho, 2020-05-21:
> in Zadig, I listed all devices, selected the Nikon driver and replaced it by libusbk.

The purpose of usbdk is to _avoid_ replacing the kernel driver.

Additionally, if you were to replace the kernel driver, the WinUSB driver would generally be a better choice.

---

> On attempting to set the option, it returns -5 which I believe is

If you're indeed getting -5 from `libsusb_set_option`, then there's a (at least a documentation) bug in libusb: that function is only [supposed](https://github.com/libusb/libusb/blob/fef88bf2b91ee4d7cfcc9d3b799a2996ecfe0d2c/libusb/libusb.h) to return either:

 - `LIBUSB_SUCCESS = 0`,
 - `LIBUSB_ERROR_INVALID_PARAM = -2`
 - or `LIBUSB_ERROR_NOT_SUPPORTED = -12`.

Try running your test again passing the `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` environment variables.

RyanHope, 2020-05-21:
See my previous comment, I already explained why this happens and the fix.

jonasmalacofilho, 2020-05-22:
> See my previous comment, I already explained why this happens and the fix.

@RyanHope 

Thanks for chipping in, but you previously reported va_list + ctypes problems only on 32-bit environments, and @jonoboy9 claims to be running 64-bit Python and libusb.

> The va_list (...) aspect of the libusb_set_option is not really supported by ctypes. **Not sure why this works on 64bit** but 32bit definitely doesn't like it.

Additionally, the code bellow just fine on Linux/x86_64, so ctypes is at least able to handle libsub_set_option _sometimes:_

```
>>> from ctypes import *
>>> import usb.backend.libusb1
>>> backend.lib.libusb_set_option.argtypes = [c_void_p, c_int]
>>> backend.lib.libusb_set_option(backend.ctx, 1)
-12 # expected, as usbdk obviously isn't supported on Linux
```

jonoboy9, 2020-05-22:
The purpose of usbdk is to avoid replacing the kernel driver.
@jonasmalacofilho Should I delete the libusbK driver so as to reinstall the previous driver. I am not too strong with device divers.

Thanks for your help especially for the debug variables..
I have found my problem. usbdk never was installed properly. No wonder I got -5:
libusb: error [windows_set_option] UsbDk backend not available

jonasmalacofilho, 2020-05-22:
Related: libusb/libusb#735 ("Document that libusb_set_option may also return NOT_FOUND")

jonasmalacofilho, 2020-05-22:
@jonoboy9 

> Should I delete the libusbK driver so as to reinstall the previous driver. I am not too strong with device divers.

I'm honestly not very familiar with UsbDk, so I'm not sure whether it's compatible with the libusbK driver.  Still, it's probably better to revert to the stock driver if (with UsbDk) a different one wont be _necessary.¹_

You might also be able to skip UsbDk altogether if your device already uses the WinUSB or Microsoft Generic HID drivers, or if you manually set it to use WinUSB (with Zadig).  Even libusbK _could²_ work, without `libusb_set_option(..., 1)`.

_¹ I've seen at least one issue caused libusbK when used in conjunction with libusb._  
_² See ¹._

RyanHope, 2020-05-22:
Well, I was able to identify 1 case where it worked and one where it couldn't. In my testing it seemed to be related to a 32bit / 64bit difference but maybe it was also connected to python version. Not 100% sure. What I do know is that I have a patched version of libusb that replaced the va arg with a fixed int option that I use with an option specific to turning on usbdk. I then patch pyusb to support this option as well because duck type modifying the signature at run time does not seem to work well either.

mcuee, 2020-07-01:
FYI, usbdk is a different backend for Windows USB. The main benefit is that it is a filter driver so that existing driver is maintained.

If the existing driver is already supported by libusb Windows (HID driver, WinUSB driver, libusbk.sys or libusb0.sys), then there is no need to use usbdk.

copslock, 2021-05-05:
the libusb0 in libusbk is an wrapper for libusbK

jonasmalacofilho, 2021-05-05:
@copslock 

Since libusb0, libusbk and libusbK are extensively overloaded terms, I unfortunately have no idea what you meant.

Also, do you have a reference you can link to?

copslock, 2021-05-06:
> @copslock
> 
> Since libusb0, libusbk and libusbK are extensively overloaded terms, I unfortunately have no idea what you meant.
> 
> Also, do you have a reference you can link to?

You can download the libusbK from sourceforge,not the libusb or libusb-win32 ones,both are not related with libusbK at all,the libusbK wizard will try to install libusbK driver for your USB devices in Windows and copy the libusb0.dll into system32,which is called by the pyusb libusb0.py backend.Most of the anwsers on the stackoverflow are foucused on "how to properly use the libusb1 API",which is impossible for some specific API that not implemented on Windows platform at all and it returns the LIBUSB_ERROR_NOT_FOUND error ,because you should not using the libusb1.0 API at all,the libusb0 dll already provided the wrapper API for libusbK.The root cause of this problem is the pyusb backend will always prefer to use the libusb1 than libusb0 which means you may inciddently loaded the libusb1 dll from some other program's folder(which is in Windows PATH),and it won't try to load the correct libusb0 dll,you can find this logic in pyusb debug log output.Many guys on the internet did't find this logic problem,instead wasting time on fixing the problem in libusb1

copslock, 2021-05-06:
I think it's a good idea to add an option for pyusb backend version force-selcetion.I had already seen **a lot** of complain caused by the libusb version selecting schema.

jonasmalacofilho, 2021-05-06:
@copslock 

> [...]

There are also reasons for _not_ installing the libusbK driver, and instead use WInUSB or even the Generic HID Driver on Windows.  As there are reasons for not using a deprecated and unmaintained API like libusb0.  Fun fact: on some non-Windows platforms, libusb0 is now provided by a [backward-compatibility shim](https://github.com/libusb/libusb-compat-0.1) implemented with libusb1 (e.g. [ArchLinux](https://archlinux.org/packages/extra/x86_64/libusb-compat/)).

> you may inciddently loaded the libusb1 dll from some other program's folder(which is in Windows PATH),

DLL search paths are not the same as executable search paths (i.e. `PATH`).  Which makes loading libusb1.dll from _another program's folder_ very unlikely.

> instead wasting time on fixing the problem in libusb1

Improving libusb1 (in the cases where that's necessary) is exactly what we should encourage, since (AFAIK) that's the only currently maintained cross-platform USB API.

> I think it's a good idea to add an option for pyusb backend version force-selcetion.

[It's already possible](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-do-i-enforce-a-backend) to force the use of a specific backend on PyUSB.  [As is](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-can-i-pass-the-libusb-library-path-to-the-backend) using a custom DLL finding function for any given backend.

mcuee, 2021-05-07:
Recommendation is to use WinUSB from libusb project unless for HID device. 

libusb FAQ:
https://github.com/libusb/libusb/wiki/FAQ
https://github.com/libusb/libusb/wiki/FAQ#How_to_use_libusb_under_Windows

mcuee, 2021-06-15:
What is the current recommended workaround for this issue if they want to use usbdk backend? 

> [It's already possible](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-do-i-enforce-a-backend) to force the use of a specific backend on PyUSB.  [As [...]

jonasmalacofilho, 2021-06-15:
> @jonasmalacofilho What will be the current recommended workaround for this issue if they want to use usbdk backend?

Assuming what is required by libusb is for `LIBUSB_OPTION_USE_USBDK` to be set, it should be sufficient to call

```
>>> from ctypes import *
>>> import usb.backend.libusb1
>>> backend.lib.libusb_set_option.argtypes = [c_void_p, c_int]
>>> backend.lib.libusb_set_option(backend.ctx, 1)
```

before proceeding to find the device (perhaps forcing the use of the `libusb1` backend, per our FAQ (linked above).

This was mentioned earlier in this thread, and there was some dispute over a possible ctypes limitation.  But I couldn't reproduce that problem, and it also didn't appear to affect modern Python installations.

mcuee, 2021-06-15:
I see. Maybe this can be added to the tutorial or FAQ.

copslock, 2021-07-25:
@jonasmalacofilho 
I discovered such issue with the github.com/scat tool on Windows plarform,it works fine with legacy libusb0 backend,but output nothing(I guess due to the newer libusb1 API incompatiable with the libusbK kernel driver API interface).
The key point of this problem is the pyusb backend python script performs the DLL searching in every Windows PATH directory,and it will prefer to load the libusb1 than libusb0 DLL,that's what cause the problem to my scat program,I don't know the behaviour in other programs but at least in my one.That's why I said version selection option should be added to the python script.Believe it or not,You can check this out by setting PYUSB_DEBUG=all to see the searching log output.
I was just wrriten down in case of anyone else going into the same paradox like me before

mcuee, 2021-07-25:
@copslock Where is this github.com/scat tool? It points to no github repo.

mcuee, 2021-07-25:
Is it this one: https://github.com/fgsect/scat ?

mcuee, 2021-07-25:
> I discovered such issue with the github.com/scat tool on Windows plarform,it works fine with legacy libusb0 backend,but output nothing with libusb1 DLL(I guess due to the newer libusb1 API incompatiable with the libusbK kernel driver API interface).
> The key point of this problem is the pyusb backend python script performs the DLL searching in every Windows PATH directory, and it will prefer to load the libusb1 than libusb0 DLL,that's what cause the problem to my scat program, I don't know the behaviour in other programs but at least in my one. That's why I said version selection option should be added to the python script.Believe it or not, You can check this out by setting PYUSB_DEBUG=all to see the searching log output.
> I was just wrriten down in case of anyone else going into the same paradox like me before

You seem to have a lot of misunderstandings of the relation between libusb-1.0, libusb-win32 (libusb-0.1 API on Windows and with extensions) and libusbk (Windows API). 

Please read the FAQ here and hopefully it will help.
https://github.com/libusb/libusb/wiki/FAQ#libusbwin32_libusbK_and_libusb_project

Zadig or libusb-inf-wizard will install WinUSB/libusbk.sys/libusb0.sys driver along with libusb0.dll and libusbk.dll. So yes the libusb-0.1 backend of pyusb will work out of the box once you use Zadig or libusbk-inf-wizard to install the supported driver.

This is not the case with libusb-1.0.dll since it is not installed. There is a reason behind it, libusb0.dll and libusbk.dll are pretty stable already (okay, we have two new libusbk release recently). libusb-1.0.dll is still under changes pretty often.

libusb-1.0.dll is supposed to work with device with WinUSB (recommended) or libusbk.sys (working well in 1.0.23 or latest git) or libusb0.sys (not stable, do not use).

libusb0.dll is working with libusbk.sys and libusb0.sys.

It is recommended you move to libusb-1.0 API and not sticking to the old libusb-0.1 API.

mcuee, 2021-07-25:
> This was mentioned earlier in this thread, and there was some dispute over a possible ctypes limitation. But I couldn't reproduce that problem, and it also didn't appear to affect modern Python installations.

I see. It seems that we can probably improve the documentation and I think this is a non issue, unless people can demonstrate the issue with supported Python version, then they can create a new issue.

copslock, 2021-07-25:
@mcuee 
the relationship between libusbK&libusb is mystery,I don't have enough time to find what the hell is going on with them,Indeed,we should prefer to use newer version but it doesn't work at least in current point.So I have to use libusb0.Yes,it's the fgsec/scat,I forgot about the author in the path.If you are interested in this issue,you can locate the root cause of libusb1 issue with scat pyusb backend.and try to improve it.
The main reason I wrote this feedback is to remind the later guys that don't walk into the libusb1 dead loop as you can see a lot of Google results suggest only "how to use libusb1",no one realized the root cause behind it

copslock, 2021-07-25:
`'NotImplementedError: Operation not supported or unimplemented on this platform`
This issue is the most famous search word of libusb1 issue,almost all of them ran into the libusb1 paradox

mcuee, 2021-07-25:
>Yes,it's the fgsec/scat,I forgot about the author in the path.If you are interested in this issue,you can locate the root cause of libusb1 issue with scat pyusb backend.and try to improve it.

Please point out the exact issue from the above repo. Thanks. Mostly it will be related to documentation.
https://github.com/fgsect/scat/issues

mcuee, 2021-07-25:
> `'NotImplementedError: Operation not supported or unimplemented on this platform`
> This issue is the most famous search word of libusb1 issue,almost all of them ran into the libusb1 paradox

I belive the FAQ documentation will help. If you think this FAQ entry is not good enough, please create pull request to improve it.
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst

copslock, 2021-07-25:
@mcuee 
I mean the 
`'NotImplementedError: Operation not supported or unimplemented on this platform`
error

jonasmalacofilho, 2021-07-25:
I'm working on adding a new API to get a USBDK-enabled libusb1 backend.  The original plan was just to document how to set it up manually, but it seems that only one mode of operation is possible for a given libusb1 context.

(@mcuee, I sent you a few more details by email) 

---

In the meantime, I have temporarily locked this discussion.

@copslock,

While you appear to have good intentions, you're also creating even more confusion.

First, this issue relates to enabling USBDK support _on the_ libusb1 backend.

Next...  Yes, `NotImplementedError` errors are very common, but as the maintainer of this project (and, thus, one of people that frequently answers questions about those errors), there are also caused by many different things, almost none of which have as the optimal solution just using libusb0.

Additionally, the feature which you continue to ask for in PyUSB already exists, and has existed for a long time.  As I mentioned [back in May](https://github.com/pyusb/pyusb/issues/200#issuecomment-833239645):

> [It's already possible](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-do-i-enforce-a-backend) to force the use of a specific backend on PyUSB. [As [...]

Finally, please reconsider the "I don't have enough time to find _what the hell is going on_" attitude, especially when interacting with maintainers of open-source projects.  (But I would argue that it's not a productive attitude in any context).  If you don't have time to understand something, that's fine, but value those that do understand it, their time, and what they tell you.

mcuee, 2021-08-03:
It seems to me the following codes will activate usbdk backend. Unfortunately usbdk does not work for my on my Windows laptop with the FX3 device (it becomes an unknown "usbdk device" with no driver attached once I try to get active configuration or set_configuration.

```
 python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> from usb.backend import libusb1
>>> import usb.util
>>> be = libusb1.get_backend()
>>> be
<usb.backend.libusb1._LibUSB object at 0x0000018995CFDA00>
>>> from ctypes import *
>>> be.lib.libusb_set_option.argtypes = [c_void_p, c_int]
>>> be.lib.libusb_set_option(be.ctx, 1)
0
>>> cfg = dev.get_active_configuration()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\work\mypy39x64\lib\site-packages\usb\core.py", line 921, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\work\mypy39x64\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\work\mypy39x64\lib\site-packages\usb\core.py", line 255, in get_active_configuration
    raise USBError('Configuration not set')
usb.core.USBError: [Errno None] Configuration not set
>>> dev
<DEVICE ID 04b4:00f3 on Bus 001 Address 003>
>>> dev.set_configuration(1)
>>> dev
<DEVICE ID 04b4:00f3 on Bus 001 Address 003>
```

jonasmalacofilho, 2021-08-03:
@mcuee,

On the PyUSB side, please test with #381 and `be = libusb1.get_usbdk_backend()`.

(Note: the API design is not final).

mcuee, 2021-08-04:
@jonasmalacofilho It seems to work. Issue is the same as above, after ```cfg = dev.get_active_configuration()```, the device becomes an unknown "usbdk device" with no driver loaded.

```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> from usb.backend import libusb1
>>> be = libusb1.get_backend()
>>> be = libusb1.get_usbdk_backend()
>>> be
<usb.backend.libusb1._LibUSB object at 0x0000025514BD87F0>
>>> dev = usb.core.find(idVendor = 0x04b4, idProduct = 0x00f3, backend=be)
>>> dev
<DEVICE ID 04b4:00f3 on Bus 001 Address 002>
>>> cfg = dev.get_active_configuration()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 921, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 255, in get_active_configuration
    raise USBError('Configuration not set')
usb.core.USBError: [Errno None] Configuration not set
```

[Note: device becomes an "Other devices -- UsbDk device"]

Device USB\Vid_2B23&Pid_CAFE&Rev_0001\UsbDk0 was configured.

Driver Name: null
Class Guid: {00000000-0000-0000-0000-000000000000}
Driver Date: 
Driver Version: 
Driver Provider: 
Driver Section: 
Driver Rank: 0x0
Matching Device Id: 
Outranked Drivers: 
Device Updated: false
Parent Device: USB\ROOT_HUB30\4&36020d6f&0&0

mcuee, 2021-08-04:
Looks like usbdk itself does not work under my Windows machine. But pyusb itself is okay.

```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> from usb.backend import libusb1
>>> be = libusb1.get_usbdk_backend()
>>> be
<usb.backend.libusb1._LibUSB object at 0x000001E4B6BD2370>
>>> dev = usb.core.find(idVendor = 0x04b4, idProduct = 0x00f3, backend=be)
>>> dev.set_configuration()
```

The device becomes a faulty USB device, because "USB Device Descriptor Request Failed".

So it does not seem to be an issue with pyusb here, again, it is rather a usbdk issue on my Windows machine. I will try another device and get libusb C code works first, then I will try pyusb again.

mcuee, 2021-08-04:
Hmm, libusb works fine with usbdk. So there is an issue with pyusb (master or the usbdk branch).

```
C:\work\libusb\libusb [master ≡ +1 ~1 -0 !]> git diff
diff --git a/examples/xusb.c b/examples/xusb.c
index 61ce4dd..7cb8cfb 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -1110,7 +1110,7 @@ int main(int argc, char** argv)
        r = libusb_init(NULL);
        if (r < 0)
                return r;
-
+       libusb_set_option(NULL, LIBUSB_OPTION_USE_USBDK);
        // If not set externally, and no debug option was given, use info log level
        if ((old_dbg_str == NULL) && (!debug_mode))
                libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_INFO);
C:\work\libusb\libusb [master ≡ +1 ~1 -0 !]> .\x64\Release\examples\xusb.exe 04b4:00f3
Using libusb v1.0.24.11647

Opening device 04B4:00F3...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 04B4:00F3
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.00.00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Cypress"
   String (0x02): "WestBridge "
   String (0x03): "0000000004BE"

Releasing interface 0...
Closing device...
```

mcuee, 2021-08-04:
@jonasmalacofilho 

Do we set up the usbdk option immediately after libusb_init()?

https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga07d4ec54cf575d672ba94c72b3c0de7c
```
LIBUSB_OPTION_USE_USBDK 	
Use the UsbDk backend for a specific context, if available.

This option should be set immediately after calling libusb_init(), otherwise unspecified behavior may occur.

Only valid on Windows.
```

mcuee, 2021-08-04:
The FX3 bootloader device does not work too well with usbdk under Windows, but at least xusb can talk it once. 

usbdk works by replacing the original driver temporally during the access, then detach itself once the access is closed so the original driver comes back after the program exits. The FX3 device may become problematic during the detach process.

However, I tried another device (an old iAR J-Link v3 JTAG debugger), xusb works fine with the device, and pyusb does not. So I think there is an issue with pyusb here.

mcuee, 2021-08-04:
Debug log for xusb

<details>
<summary> click to expand </summary>

```
C:\work\libusb\libusb\x64\Release [master ≡ +1 ~1 -0 !]> .\examples\xusb.exe 1366:0101
Using libusb v1.0.24.11647

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000410] [00005458] libusb: debug [libusb_init] created default context
[ 0.000658] [00005458] libusb: debug [libusb_init] libusb v1.0.24.11647
[ 0.000898] [00005458] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.001135] [00005458] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001353] [00005458] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.001685] [00005458] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001922] [00005458] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010675] [00005458] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.016313] [00005458] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.8.0
[ 0.025024] [00005458] libusb: debug [windows_init] UsbDk backend is available
[ 0.025363] [00005458] libusb: debug [windows_set_option] switching context 0000016AA60CF5F0 to use UsbDk backend
Opening device 1366:0101...
[ 0.025681] [00005458] libusb: debug [libusb_get_device_list]
[ 0.026043] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026194] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026324] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026481] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026559] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026667] [00005458] libusb: debug [libusb_get_device_descriptor]
[ 0.026777] [00005458] libusb: debug [libusb_open] open 1.2
[ 0.026502] [0000177c] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.304627] [00005458] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.304801] [00005458] libusb: debug [libusb_unref_device] destroy device 1.11
[ 0.304924] [00005458] libusb: debug [libusb_unref_device] destroy device 1.12
[ 0.305023] [00005458] libusb: debug [libusb_unref_device] destroy device 1.15
[ 0.305136] [00005458] libusb: debug [libusb_unref_device] destroy device 1.3

Reading device descriptor:
[ 0.305394] [00005458] libusb: debug [libusb_get_device_descriptor]
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 1366:0101
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: [ 0.305975] [00005458] libusb: debug [libusb_submit_transfer] transfer 0000016AA6176F28
[ 0.306069] [00005458] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.306358] [00005458] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.306475] [00005458] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.306614] [0000177c] libusb: debug [windows_iocp_thread] transfer 0000016AA6176F28 completed, length 16
[ 0.306626] [00005458] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.306838] [00005458] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.306903] [00005458] libusb: debug [handle_event_trigger] event triggered
[ 0.306972] [00005458] libusb: debug [windows_handle_transfer_completion] handling transfer 0000016AA6176F28 completion with errcode 0, length 16
[ 0.307084] [00005458] libusb: debug [usbd_status_to_libusb_transfer_status] USBD_STATUS 0x80000800 translated to LIBUSB_TRANSFER_ERROR
[ 0.307196] [00005458] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.307287] [00005458] libusb: debug [usbi_handle_transfer_completion] transfer 0000016AA6176F28 has callback 00007FF65CBD8830
[ 0.307399] [00005458] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.307487] [00005458] libusb: debug [libusb_free_transfer] transfer 0000016AA6176F28
[ 0.307543] [00005458] libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
[ 0.307706] [00005458] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.307758] [00005458] libusb: warning [raw_desc_to_config] still 7 bytes of descriptor data left
             nb interfaces: 1
              interface[0]: id = 0
[... 93 lines cut ...]
[ 0.313789] [00005458] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.313840] [00005458] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.313936] [0000177c] libusb: debug [windows_iocp_thread] transfer 0000016AA6176F28 completed, length 16
[ 0.314028] [00005458] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.314122] [00005458] libusb: debug [handle_event_trigger] event triggered
[ 0.314177] [00005458] libusb: debug [windows_handle_transfer_completion] handling transfer 0000016AA6176F28 completion with errcode 0, length 16
[ 0.314251] [00005458] libusb: debug [usbd_status_to_libusb_transfer_status] USBD_STATUS 0x80000800 translated to LIBUSB_TRANSFER_ERROR
[ 0.314323] [00005458] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.314374] [00005458] libusb: debug [usbi_handle_transfer_completion] transfer 0000016AA6176F28 has callback 00007FF65CBD8830
[... cut ...]
[ 0.314734] [00005458] libusb: debug [libusb_close]
[ 0.339100] [00005458] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.339250] [00005458] libusb: debug [libusb_exit] destroying default context
[ 0.339379] [0000177c] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.342268] [00005458] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.342389] [00005458] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088

```

</details>

mcuee, 2021-08-04:
pyusb debug log.

```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> from usb.backend import libusb1
>>> be = libusb1.get_usbdk_backend()
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000292] [00001770] libusb: debug [libusb_init] libusb v1.0.24.11647
[ 0.000472] [00001770] libusb: debug [usbi_add_event_source] add HANDLE 00000000000001A4 events 0
[ 0.000648] [00001770] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000827] [00001770] libusb: debug [usbi_add_event_source] add HANDLE 00000000000001A8 events 0
[ 0.001031] [00001770] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.001199] [00001770] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003496] [00001770] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.004413] [00001770] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.8.0
[ 0.009810] [00001770] libusb: debug [windows_init] UsbDk backend is available
[ 0.010140] [00001770] libusb: debug [windows_set_option] switching context 0000019F37BFE670 to use UsbDk backend
>>> [ 0.011139] [00002814] libusb: debug [windows_iocp_thread] I/O completion thread started
be
<usb.backend.libusb1._LibUSB object at 0x0000019F3981D220>
>>> dev = usb.core.find(idVendor = 0x1366, idProduct = 0x0101, backend=be)
[62.677827] [00001770] libusb: debug [libusb_get_device_list]
[62.678281] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.678516] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.678714] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.678916] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.679100] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.679256] [00001770] libusb: debug [libusb_get_device_descriptor]
[62.679399] [00001770] libusb: debug [libusb_unref_device] destroy device 1.4
[62.679501] [00001770] libusb: debug [libusb_unref_device] destroy device 1.11
[62.679556] [00001770] libusb: debug [libusb_unref_device] destroy device 1.12
[62.679607] [00001770] libusb: debug [libusb_unref_device] destroy device 1.15
[62.679657] [00001770] libusb: debug [libusb_unref_device] destroy device 1.3
>>> dev
<DEVICE ID 1366:0101 on Bus 001 Address 002>
>>> cfg = dev.get_active_configuration()
[104.726062] [00001770] libusb: debug [libusb_open] open 1.2
[105.006327] [00001770] libusb: debug [libusb_get_configuration]
[105.006463] [00001770] libusb: debug [libusb_get_configuration] active config 0
[105.006649] [00001770] libusb: debug [libusb_get_config_descriptor] index 0
[105.006792] [00001770] libusb: warning [raw_desc_to_config] still 7 bytes of descriptor data left
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 921, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 255, in get_active_configuration
    raise USBError('Configuration not set')
usb.core.USBError: [Errno None] Configuration not set
```

mcuee, 2021-08-04:
pyusb debug log.

```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.environ['PYUSB_DEBUG'] = 'debug'
>>> import usb.core
>>> import usb.util
>>> from usb.backend import libusb1
>>> be = libusb1.get_usbdk_backend()
2021-08-04 15:40:49,751 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7ffbcd3d0000 at 0x1a4adb9e610>)
>>> dev = usb.core.find(idVendor=0x1366, idProduct=0x0101, backend = be)
2021-08-04 15:41:14,560 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2021-08-04 15:41:14,561 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD880>)
2021-08-04 15:41:14,561 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD6D0>)
2021-08-04 15:41:14,562 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD820>)
2021-08-04 15:41:14,562 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD670>)
2021-08-04 15:41:14,562 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD790>)
2021-08-04 15:41:14,562 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD550>)
>>> dev
<DEVICE ID 1366:0101 on Bus 001 Address 002>
>>> dev.set_configuration(1)
2021-08-04 15:42:06,557 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD550>, 0)
2021-08-04 15:42:06,558 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001A4AFEFD550>)
2021-08-04 15:42:06,831 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x000001A4AE3CC3A0>, 1)
>>> cfg = dev.get_active_configuration()
2021-08-04 15:42:46,319 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A4AFEFD550>, 0)
>>> cfg
<CONFIGURATION 1: 100 mA>
```

mcuee, 2021-08-04:
Both PYUSB and LIBUSB debug log are included here.
```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.environ['PYUSB_DEBUG'] = 'debug'
>>> import usb.core
>>> import usb.util
>>> from usb.backend import libusb1
>>> be = libusb1.get_usbdk_backend()
2021-08-04 16:12:34,748 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7ffbef5b0000 at 0x1d3a9e6e5e0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000253] [000050fc] libusb: debug [libusb_init] libusb v1.0.24.11647
[ 0.000385] [000050fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000134 events 0
[ 0.000539] [000050fc] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000699] [000050fc] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000130 events 0
[ 0.000844] [000050fc] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000965] [000050fc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003160] [000050fc] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.004065] [000050fc] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.8.0
[ 0.009487] [000050fc] libusb: debug [windows_init] UsbDk backend is available
[ 0.009824] [000050fc] libusb: debug [windows_set_option] switching context 000001D3AA528090 to use UsbDk backend
>>> [ 0.010940] [00000280] libusb: debug [windows_iocp_thread] I/O completion thread started

>>> dev = usb.core.find(idVendor = 0x1366, idProduct = 0x0101, backend=be)
2021-08-04 16:13:05,565 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[30.817371] [000050fc] libusb: debug [libusb_get_device_list]
2021-08-04 16:13:05,567 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C56D0>)
[30.818110] [000050fc] libusb: debug [libusb_get_device_descriptor]
2021-08-04 16:13:05,567 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C5520>)
[30.818536] [000050fc] libusb: debug [libusb_get_device_descriptor]
2021-08-04 16:13:05,568 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C5670>)
[30.818871] [000050fc] libusb: debug [libusb_get_device_descriptor]
2021-08-04 16:13:05,568 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C5400>)
[30.819187] [000050fc] libusb: debug [libusb_get_device_descriptor]
2021-08-04 16:13:05,568 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C55E0>)
[30.819480] [000050fc] libusb: debug [libusb_get_device_descriptor]
2021-08-04 16:13:05,569 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C53A0>)
[30.819777] [000050fc] libusb: debug [libusb_get_device_descriptor]
[30.819903] [000050fc] libusb: debug [libusb_unref_device] destroy device 1.4
[30.820003] [000050fc] libusb: debug [libusb_unref_device] destroy device 1.11
[30.820096] [000050fc] libusb: debug [libusb_unref_device] destroy device 1.12
[30.820192] [000050fc] libusb: debug [libusb_unref_device] destroy device 1.15
[30.820288] [000050fc] libusb: debug [libusb_unref_device] destroy device 1.3
>>> dev
<DEVICE ID 1366:0101 on Bus 001 Address 002>
>>> cfg = dev.get_active_configuration()
2021-08-04 16:13:24,093 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001D3AC1C53A0>)
[49.343839] [000050fc] libusb: debug [libusb_open] open 1.2
2021-08-04 16:13:24,377 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration(<usb.backend.libusb1._DeviceHandle object at 0x000001D3AA69C3D0>)
[49.628656] [000050fc] libusb: debug [libusb_get_configuration]
[49.628745] [000050fc] libusb: debug [libusb_get_configuration] active config 0
2021-08-04 16:13:24,378 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D3AC1C53A0>, 0)
[49.629167] [000050fc] libusb: debug [libusb_get_config_descriptor] index 0
[49.629246] [000050fc] libusb: warning [raw_desc_to_config] still 7 bytes of descriptor data left
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 921, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\work\libusb\pyusb_usbdk\usb\core.py", line 255, in get_active_configuration
    raise USBError('Configuration not set')
usb.core.USBError: [Errno None] Configuration not set
```

mcuee, 2021-08-09:
@RyanHope This must work before. Just wondering which usbdk version, libusb version and pyusb version you have made usbdk working. Thanks.

mcuee, 2021-08-09:
@abulka same question to you, just wonder which version of usbdk/libusb/pyusb you have when you got usbdk to work, and whether they are still working now or not. Thanks.

mcuee, 2021-08-09:
Apparently it is not libusb version, as I have even tested back to libusb-1.0.21 where I need to use compile time option to enable usbdk and it has the same issue (without the need to use ctypes since that libusb-1.0.dll is fixed to used usbdk).

mcuee, 2021-08-09:
And it is also not like a pyusb issue as I went back to pyusb-1.0.2 and it has the same issue.

So looks like there is a change in usbdk.

mcuee, 2021-08-09:
Switching back to usbdk 1.0.21 version (dated Apr 30,2019) does not help either.

mcuee, 2021-08-09:
Unlocked this conversation to allow users who have successfully using usbdk to chime in.

mcuee, 2021-08-20:
@jonasmalacofilho Finally figured out a possible solution. We need to check usbdk backend, if using usbdk backend, we should not call self.backend.set_configuration(self.handle, cfg.bConfigurationValue).

The following is a quick hack to demonstrate this. And I tested it to work properly.

```

(py39venv) C:\work\libusb\pyusb_usbdk [add-support-for-usbdk ≡ +1 ~1 -0 !]> git diff
diff --git a/usb/core.py b/usb/core.py
index 9a63fb8..46984a5 100644
--- a/usb/core.py
+++ b/usb/core.py
@@ -156,7 +156,7 @@ class _ResourceManager(object):
             raise ValueError("Invalid configuration " + str(config))

         self.managed_open()
-        self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
+        #self.backend.set_configuration(self.handle, cfg.bConfigurationValue)

         # cache the index instead of the object to avoid cyclic references
         # of the device and Configuration (Device tracks the _ResourceManager,
```

Test application (not that proper, but get the job done as a test).

[11 more replies]

closed 2024-04-30 by jonasmalacofilho