pyusb/pyusb #206: BSOD when executing script with usb.core

anffray, 2018-07-23:
Hello ! 
I am programming in Python a code which need to interact with the usb ports. Thus I need to use functions from the usb library and more specifically the usb.core class.

The problem is that whenever I try to execute just simple lines such as these :

```
import usb
usb.core.find(find_all = True)
```
Windows crashes with a BSOD because of a WDF violation error...
I tried to reinstall pysub using  

`pip install --upgrade --force-reinstall pyusb`

 but it doesn't improve anything...

Thank you very much for the help you will provide me ! :)

praveen1013, 2018-07-24:
Try installing pycharm and debug this code and step into the find function to figure out where exactly in the code it results in bluescreen

mcuee, 2018-08-01:
I have never seen this problem on my side. Please make sure you have the latest version of libusbK/libusb-win32 if you have them installed.

anffray, 2018-08-01:
Hi all and thank you for your answers,
I reinstalled libusb using these instructions and it works for now : https://stackoverflow.com/questions/33972145/pyusb-on-windows-8-1-no-backend-available-how-to-install-libusb

closed 2018-08-01 by anffray