pyusb/pyusb #207: usb.core.USBError: [Errno 32] Pipe error

ysharshit, 2018-08-03:
I am trying to read/write using a USB-OTG port in Altera De0 Nano SoC. I have the Atlas SoC installed in a SD card for using the Arm Linux.
The Descriptors of the device are as follows:
CONFIGURATION 1: 2 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0xa4 (164 bytes)
   bNumInterfaces       :    0x5
   bConfigurationValue  :    0x1
   iConfiguration       :    0x6 Multifunction with RNDIS
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :    0x1 (2 mA)
    INTERFACE 0: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :   0xff
     iInterface         :    0x8 RNDIS Communications Control
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x9
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x9 RNDIS Ethernet Data
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
    INTERFACE 2: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x2
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x1
     iInterface         :    0xc CDC Abstract Control Model (ACM)
      ENDPOINT 0x84: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x84 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0xa (10 bytes)
       bInterval        :    0x9
    INTERFACE 3: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x3
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0xd CDC ACM Data
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
    INTERFACE 4: Mass Storage ==============================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x4
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x8 Mass Storage
     bInterfaceSubClass :    0x6
     bInterfaceProtocol :   0x50
     iInterface         :    0x1 Mass Storage
      ENDPOINT 0x85: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x85 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x3: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x1

I have removed the default drivers which get loaded by the device by using the sudo rmmod command. The drivers removed by me are: cdc_acm, rndis_wlan, rndis_host. I am using the device as USB Mass Storage for Bulk Transfer (Interface 4)
I am running the following python code:
``` import sys, usb.core, time
#For USB OTG Hexadecimal VendorID=0x1d6b & ProductID=0x104
dev = usb.core.find(idVendor=0x1d6b, idProduct=0x104) 
interface=4
if dev is None:
    sys.exit("No Panic button found in the system");

try:
    if dev.is_kernel_driver_active(interface) is True:
        dev.detach_kernel_driver(interface)
    usb.util.claim_interface(dev,interface)
	usb.util.release_interface(dev, interface)
	dev.attach_kernel_driver(interface)
except usb.core.USBError as e:
    sys.exit("Kernel driver won't give up control over device: %s" % str(e))

try:
    dev.set_configuration()
    dev.reset()
except usb.core.USBError as e:
    sys.exit("Cannot set configuration the device: %s" % str(e))
print dev
endpoint_in = dev[0][(interface,0)][0]
endpoint_out = dev[0][(interface,0)][1]
print endpoint_in
print endpoint_out
msg = 'test123'
data1 = dev.write(endpoint_out, msg,100)
print data1
data = dev.read(endpoint_in, len(msg),100)
sret = ''.join([chr(x) for x in data])
print sret
print repr(data.tostring())
usb.util.dispose_resources(dev)
```
I am able to write to the device using "data1 = dev.write(endpoint_out, msg,100)" but when I try reading from the device "data = dev.read(endpoint_in, len(msg),100)", I am getting the following error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

I tried to search for this issue but could not find much.
In [http://christopherpeplin.com/2012/02/bulk-usb-throughput](url), the following is mentioned:

> The complete fix for OpenXC’s use case will be to make sure every message is padded out to 64 bytes and to request bulk transfer sizes that are big enough to get good throughput but small enough to not be too delayed. I have yet to confirm, but my understanding is that if we request a 4KB read and don’t mark the end of the transfer with a less-than-max length packet, the host device will block waiting for the rest of the requested data.

I tried to send a 64 byte data using the following string:
abcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcdabcd
The write command shows 64 bytes written but the same error 32 arises when I try to read from the device. I am new to USB programming and I am stuck up here. Can somebody please help?

mcuee, 2018-08-04:
You need to understand the communication protocol in order to use libusb/pyusb. USB mass storage device is not a good start to use libusb/pyusb. You can read the libsb faq here.
https://github.com/libusb/libusb/wiki/FAQ#Can_I_use_libusb_to_open_a_file_on_an_USB_storage_device

mcuee, 2018-08-05:
What is your objective? If your objective is to use pyusb and libusb, then this is not a good device to start with. pyusb/libusb are more suitable for generic device not fall into specific class (eg: RNDIS, CDC-ACM and USB Mass Storage).

mcuee, 2018-08-07:
As long as you know the communication protocol or you have control on your device side (it seems to me that you have control since this is a development board), then you can use the interfaces to communicate with the device, for example, CDC-ACM will be the serial port and you can use pySerial for the communication. rndis will come out as the network interface and you can use those network related python (or other language) packages for the communication.

ysharshit, 2018-08-07:
@mcuee Thanks a lot for your guidance - I got the thing working using pyserial

closed 2018-08-07 by ysharshit

mcuee, 2018-08-10:
Glad that you solved the problem and thanks for reporting back.