pyusb/pyusb #209: Support the WinUSB API as a backend.

Elemecca, 2018-08-15:
Is there any reason (other than than no one has written it yet) that the [WinUSB API][winusb] isn't supported directly as a backend? It should be possible to call it directly via ctypes like the other backends.

I'm happy to attempt the implementation myself, I just want to know before I get too far into the project if there's anything significant that's prevented it being done before now.

[winusb]: https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb

mcuee, 2018-08-18:
FYI there is this WinUSBPy. I have not tried it though.
  https://github.com/felHR85/WinUsbPy.git

FYI also: pyusb already has the libusb-1.0 backend which supports device using WinUSB driver.

On the other hand, I think it is not a bad idea to add WinUSB backend for Windows, in that case, you do not need to have libusb-1.0.dll.

Elemecca, 2018-08-18:
Yeah, my goal in adding direct WinUSB support is to allow pyusb applications to work on Windows after just a `pip install` with no need to separately download and install libusb.

altear, 2019-01-12:
Any luck on this? 
I'm getting an `Operation not supported or unimplemented on this platform` when I try to use the libusb1 backend from Windows10

Elemecca, 2019-01-15:
@altear it's coming along slowly. It's not a high priority project, just something I'm working on in my free time.

That said, you shouldn't need my direct WinUSB support to use pyusb on Windows. The libusb1 backend should work, but it does need `libusb-1.0.dll`. Unfortunately this issue isn't the right venue for help with the libusb1 backend.

mcuee, 2021-08-10:
Main difference is that Zadig (or libusbk inf wizard) does not install libusb-1.0.dll, only libusb0.dll and libusbk.dll, so that libusb-0.1 backend will work out of the box, if you use Zadig. So you need to add libusb-1.0.dll to the right location, in order to use libusb-1.0 backend under Windows.

I do not think this is a big problem in reality.

mcuee, 2021-08-10:
https://github.com/pyusb/pyusb/blob/master/README.rst
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst

jonasmalacofilho, 2022-07-11:
> Main difference is that Zadig (or libusbk inf wizard) does not install libusb-1.0.dll [...]

It just occurred to be (while cleaning up old emails on my inbox and finding one referencing this issue) whether we know why Zadig doesn't ship `libusb-1-0.dll`, and whether that could be changed.  @mcuee, do you know?

---

A related note, for those that might not know yet: [pyocd/libusb-package](https://github.com/pyocd/libusb-package) has made the lives of developers using PyUSB a lot simpler, as it's now quite easy to ensure a suitable libusb-1.0 DLL will be available on the user's machine.[^1]

[^1]: https://github.com/pyusb/pyusb#requirements-and-platform-support

mcuee, 2022-07-11:
> A related note, for those that might not know yet: [pyocd/libusb-package](https://github.com/pyocd/libusb-package) has made the lives of developers using PyUSB a lot simpler, as it's now quite easy to ensure a suitable libusb-1.0 DLL will be available on the user's machine.

That is good to know.

BTW, python-libusb1 is also trying to ship the libusb-1.0.dll binary for Windows.
Ref: https://github.com/vpelletier/python-libusb1/blob/master/setup.py