pyusb/pyusb #210: Unknown error could anyone help me with this

adi619, 2018-08-21:
<usb.core.Device object at 0x00000135EE91DBE0>
Traceback (most recent call last):
  File "C:\Users\Downloads\test.py", line 14, in <module>
    test = dev.read(0x81,8,timeout = 50)
  File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 647, in read
    self._ctx.managed_claim_interface(self, intf)
  File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 112, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb01.py", line 462, in claim_interface
    _check(_lib.usb_claim_interface(dev_handle, intf))
  File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb01.py", line 384, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, invalid configuration 0\n'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Downloads\test.py", line 20, in <module>
    raise IOError("USB Error: %s"%str(e))
OSError: USB Error: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, invalid configuration 0\n'

mcuee, 2018-09-19:
Did you notice this " invalid configuration 0"? Your device is not configured.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee