pyusb/pyusb #211: Website link wrong repo (?)

poikilos, 2018-08-29:
This repo's Readme links to the [website on github.io] (https://pyusb.github.io/pyusb) but that site links to walac's repo. Please fix link there or write a clarification somewhere about which repo is official.

mdxs, 2019-02-25:
On a similar note: the current Homepage configured for pyusb on PyPi (see https://pypi.org/project/pyusb/) is http://walac.github.io/pyusb which results in a "404 - File not found" error and should probably be updated to https://pyusb.github.io/pyusb/

When doing so, please also make sure to use httpS links... note that the description of the project here on GitHub points to the http (non-S) variant.

mcuee, 2019-11-29:
Patches in the form of Pull Request are welcome. Thanks.

jonasmalacofilho, 2019-11-29:
Done with 47e9f49e1a334cb80da330a9274a04d9a35f8442, 28d82e256aced0fec5f92c6b6650cd6844e07a44 and eadaecd697a0b50cd9a70b97ecd3ee8b8e9fc527.

closed 2019-11-29 by jonasmalacofilho