pyusb/pyusb #212: [Errno 110] Operation timed out

EmanueleRichiedei, 2018-08-31:
Hello. i am working in this module.
I am trying to write data to a PenDrive and reading that data, but I have always a problem  ,  ia have the [Errno 110] Operation timed out

def write(intf,msg):

    ep = usb.util.find_descriptor(intf, custom_match=lambda e: usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT)
    assert ep is not None

    try:
        #scrivi dati

        dev.write(ep.bEndpointAddress, msg, 100)
        print("scrittura andata a buon fine")
    except core.USBError, r:
        print(str(r))

def read(intf):

    m = None
    ep = usb.util.find_descriptor(intf,custom_match = lambda e: usb.util.endpoint_direction(e.bEndpointAddress) ==  usb.util.ENDPOINT_IN)

    ep_in=ep.bEndpointAddress
    length = ep.wMaxPacketSize

    try:
        ret = dev.read(ep_in, length, 100)
        sret = ''.join([chr(x) for x in ret])
        assert sret == msg

    except core.USBError, r:
        print (str(r))

    return m

dev = core.find(idVendor=0x10d6)  # cerco il mio device

if dev is not  None:

    dev.reset()

    #usb.util.claim_interface(dev, 0)

    if dev.is_kernel_driver_active(0) is True:  # cotrolla se e attaccato oppure no
          dev.detach_kernel_driver(0)  # stacca il device

    dev.set_configuration()

    #get an endpoint instance
    cfg = dev.get_active_configuration()
    intf = cfg[(0,0)]

    write(intf,"ciao")

    time.sleep(1)

    print (read(intf))

    usb.util.dispose_resources(dev)
    dev.attach_kernel_driver(0)

else:
    print("NO DEVICE COLLEGATO ")
    raise ValueError("Device non found")

my output is :
ok
[Errno 110] Operation timed out
None

mcuee, 2018-09-19:
Do you know the communication protocol of the product?  In the end you still need to know the communication protocol of the device as well. How do you know your device understand "write(intf,"ciao")"?

closed 2018-09-19 by mcuee

EmanueleRichiedei, 2018-12-15:
I read the doc and i read that i can use the "write() " method with a device which use the bulk tranfer .

EmanueleRichiedei, 2018-12-15:
I am using a **pen drive** and it uses the bulk tranfer.
"write(intf,"ciao") is a functon I have wtitten to check stuff i need.
In the documentation i read i can use "dev.read()" and "dev.wtite()" while i am using bulk tranfer. 

`def write(intf,msg):

ep = usb.util.find_descriptor(intf, custom_match=lambda e: usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT)
assert ep is not None

try:
    #scrivi dati

    dev.write(ep.bEndpointAddress, msg, 100)
    print("scrittura andata a buon fine")
except core.USBError, r:
    print(str(r))`

shuvragood, 2022-09-30:
Traceback (most recent call last):
  File "C:\Users\Avinash\Music\Potentiostat_SW_A9\Potentiostat_communication.py", line 80, in usb_read
    usb_input = self.ep_in.read(size, timeout)
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\core.py", line 301, in read
    return self.device.read(self.bEndpointAddress, size, self.interface, timeout)
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\core.py", line 656, in read
    return fn(
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 481, in bulk_read
    return self.__read(_lib.usb_bulk_read,
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 566, in __read
    ret = int(_check(fn(
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 387, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'

mcuee, 2022-09-30:
DO NOT use libusb or pyusb with a USB Flash Drive or things like that. It is not meant for that purpose. You NEED to know the communication protocol of the USB devices you want to communicate with.

If not, you have to try reverse engineering.
Example: https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/