pyusb/pyusb #214: PyUSB, write to a device without detaching kernel

AIWintermuteAI, 2018-09-28:
I have a very cheap Chinese stereo camera, CHUSEI 3D. I would like to use it with ROS to obtain stereo image for the robotic arm tutorial I'm currently doing now.

Unfortunately there's no real documentation for the webcam. In default mode, if it is connected to my Ubuntu machine and I open Cheese Webcam Booth it only shows the image from one camera. On the internet I found a solution using pyusb that works for MacOs. I've tried it on my Ubuntu 16.04, but it gives me an error Error 16 : Resource is busy.

I've added dev.detach_kernel_driver(0) line to the code and ctrl_transfer worked... But this way I can't access the webcam afterwards. So, my question is, using Python and Ubuntu 16.04, is there a way I can send these commands to webcam WITHOUT detaching the kernel driver, so I still have the access to the webcam?

import cv2
import usb.core
import usb.util
import usb.backend.libusb1

cam=cv2.VideoCapture(0) 

backend = usb.backend.libusb1.get_backend(find_library=lambda x: "/usr/lib/libusb-1.0.so")   
dev = usb.core.find(idVendor=0x18e3, idProduct=0x5031,backend=backend)
dev.detach_kernel_driver(0)
# simulate the SET_CUR sequence
dev.ctrl_transfer(0x21,0x01,0x0800,0x0600,[0x50,0xff])
dev.ctrl_transfer(0x21,0x01,0x0f00,0x0600,[0x00,0xf6])
dev.ctrl_transfer(0x21,0x01,0x0800,0x0600,[0x25,0x00])
dev.ctrl_transfer(0x21,0x01,0x0800,0x0600,[0x5f,0xfe])
dev.ctrl_transfer(0x21,0x01,0x0f00,0x0600,[0x00,0x03])
dev.ctrl_transfer(0x21,0x01,0x0f00,0x0600,[0x00,0x02])
dev.ctrl_transfer(0x21,0x01,0x0f00,0x0600,[0x00,0x12])
dev.ctrl_transfer(0x21,0x01,0x0f00,0x0600,[0x00,0x04])
dev.ctrl_transfer(0x21,0x01,0x0800,0x0600,[0x76,0xc3])
k=0
print('ok')
while (k!=ord('q')):
 ret,frame=cam.read()
 frame = cv2.resize(frame,(640,480),interpolation=cv2.INTER_CUBIC)
 cv2.imshow("cam_test",frame)

 k=cv2.waitKey(18)&0xFF
 kv=k-ord('0')
 # print(kv)
 # if press 1,2,3 or 4, change the 3d camera mode
 if kv in [1,2,3,4]:
  dev.ctrl_transfer(0x21,0x01,0x0a00,0x0600,[kv,0x00])

Error message

Traceback (most recent call last):
  File "test.py", line 13, in <module>
    dev.ctrl_transfer(0x21,0x01,0x0800,0x0600,[0x50,0xff])
  File "/home/ubuntu/.local/lib/python2.7/site-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/home/ubuntu/.local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy

ixjlyons, 2018-09-28:
There is the [`attach_kernel_driver`](https://github.com/pyusb/pyusb/blob/ffe6faf42c6ad273880b0b464b9bbf44c1d4b2e9/usb/core.py#L1081) method that should work if you just need to use the device as normal after your script runs. If you want to, say, view the camera output with Cheese *while* your script is running, I'm not sure whether or not that's feasible.

AIWintermuteAI, 2018-09-28:
I think in order to use attach_kernel_driver I also need to call` usb.util.dispose_resources(dev)` 
After I do that I can attach_kernel_driver and use webcam normally, but the it seems the command I sent before using ctrl_transfer  is reset this way... It still shows only one camera image
https://www.jianshu.com/p/2ce302d588a0
Here's a link to original article where I've found the code. I've already contacted the author(fortunately I speak fluent Mandarin) and confirmed he did in fact use MacOs and not Linux.

ixjlyons, 2018-09-28:
Ah, I think I understand what you're trying to do now (use pyusb to configure the device to use stereo then exit and hope it is still configured that way). Am I right in assuming the script is actually working in that it displays the stereo feed in an OpenCV window? I suppose reattaching the kernel driver is basically just re-initializing the device back to its default operating mode. Unless the camera has a "save configuration" command or something (which would be difficult to figure out unless you have a program that uses that functionality and you can sniff the traffic), I think your options are either to use pyusb for everything whenever you need stereo image data or [write your own kernel driver](https://kernel.readthedocs.io/en/sphinx-samples/writing_usb_driver.html) so it operates that way be default.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee

TechMarvel, 2021-01-21:
Hi, I got same camera yesterday, and spent the whole day on the same problem, is there a solution now? Thanks.

AIWintermuteAI, 2021-01-21:
Nope. I just decided to use Kinect instead :)
https://www.hackster.io/dmitrywat/rgb-d-slam-with-kinect-on-raspberry-pi-4-ros-melodic-ace795

mcuee, 2021-01-21:
Basically pyusb depends on libusb, and you can not "write" to the device without detaching the kernel driver. And you re-attach the driver, it depends on the driver and the device whether your settings are kept or not. So pyusb may not be able to help you in this case.

mcuee, 2021-01-21:
If you want to use the camera the way you want, you may have to write your own kernel driver and get it accepted by the Linux kernel.

TechMarvel, 2021-01-22:
Thanks a lot for answer, I think it better stop with this camera, I will try something else. Keep going will take me to a different direction.