pyusb/pyusb #215: Pyusb-No backend available error

ChandanaVishwanatha, 2018-10-04:
I am trying to find USB Mass Storage devices attached to my PC using usb.core.find() from command prompt in windows 7. I am using Python 3.7, pyusb 1.02 and libusb1.0.22b1. After importing usb.core, I am using the above mentioned command and I am getting the following error:

File "C:...my_path....\usb\core.py", line 1263, in findusb.core.No BackendError: NO backend available

I read core.py file and understood that this is because core.py is not able to access libusb0.py, libusb1.py or openusb which are located in a folder named 'backend' which is in the same folder as core.py.

Could any one tell me if my understanding is correct or not. If yes,please suggest a way to make libusb1.py accessible from core.py.(I 've tried adding the path to environment variableand didn;t work) and please correct me if I am wrong and suggest the necessary changes.

quoidautre, 2018-11-29:
Hi,
Same error for me.
I'm on Python 3.6.6 (Windows 10) and using winpython.

Fabrice

mcuee, 2019-01-17:
You need to have libusb-1.0.dll in the Path.

quoidautre, 2019-02-13:
Hi,
Ok, thanks.

Sorry for my late response.
F.

tcarroll2, 2019-08-26:
I have install pyusb via pip and also installed libusb1 via pip and I am getting this error as well.

Mac OS Mojave 10.14.6
Python 3.7.4

Thanks for any help!

mrstock, 2019-09-10:
> I have install pyusb via pip and also installed libusb1 via pip and I am getting this error as well.
> 
> Mac OS Mojave 10.14.6
> Python 3.7.4

same issue here ... the library on mac is called `libusb-1.0.0.dylib`

Can be a name related issue ?

mrstock, 2019-09-11:
Workaround for MacOS : create a symlink named libusb-1.0.dylib linked to the right library libusb-1.0.0.dylib

```
 rwxr-xr-x   1 Stock  admin    99880 10 Set 18:49 libusb-1.0.0.dylib*
lrwxr-xr-x   1 Stock  admin       18 10 Set 19:28 libusb-1.0.dylib@ -> libusb-1.0.0.dylib
```

or as suggested by the tutorial page : 

```
>>> import usb.core
>>> import usb.backend.libusb1
>>>
>>> backend = usb.backend.libusb1.get_backend(find_library=lambda x: "/usr/lib/libusb-1.0.0.dylib")
>>> dev     = usb.core.find(..., backend=backend)
```

premananda8, 2019-09-25:
One Copy and paste this code will work.

import sys

import usb.backend.libusb1
from usb import util

path = sys.executable
path = path.replace("python.exe", "")

def connect_to_device(ser_no):
    backend = usb.backend.libusb1.get_backend(find_library=lambda
        x: path + "Lib/site-packages/libusb/_platform/_windows/x64" + "/libusb-1.0.dll")  # Lib\site-packages\libusb\_platform\_windows\x64
    print(backend)
    dev = usb.core.find(find_all=True, idVendor=your_vendorid, idProduct=your_productid, backend=backend)
    cc_dev = None
    try:
        for d in dev:
            # print(d)

            print("Serno:" + d.serial_number)
            if d.serial_number == ser_no:
                cc_dev = d

    except Exception as e:
        print(str(e))
    return cc_dev

jonasmalacofilho, 2019-09-29:
@premananda8, you forgot to mention that this requires the [libusb module](https://pypi.org/project/libusb/) to be installed.

But it's an interesting idea!

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

PhatDatPQ, 2020-09-12:
1. download the latest libusb Download [libusb](http://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/libusb-1.0.20.7z/download)

> Copy MS32\dll\libusb-1.0.dll to C:\Windows\SysWOW64

>Copy MS64\dll\libusb-1.0.dll to C:\Windows\System32

2.
>pip install libusb

This method works for me.

inio, 2020-09-15:
I just ran into this.  It turns out on mac ctypes.util.find_library doesn't search `LD_LIBRARY_PATH` and instead you need to update `DYLD_LIBRARY_PATH` to include your homebrew `/lib` directory.  Setting that environment variable correctly lets you fix the error without changing your python code.

mcuee, 2020-09-16:
Reopen this, just use for reference for the users.

jonasmalacofilho, 2020-09-16:
@mcuee 

We have a [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst), perhaps we should just link to it more often.  This also applies to your edit to the Wiki.

In fact, I think we should just disable the Wiki.  It suggests that such type of documentation is there, when it isn't (it is in `docs/faq,tutorial}.rst`), and the only other reason to have it would be for user-contributed documentation, which we don't have.

mcuee, 2020-09-17:
@jonasmalacofilho Great point. We meed mention FAQ and the Tutorial more often.

Wiki has been disabled.

closed 2020-09-17 by mcuee

mcuee, 2020-09-17:
I have also sent a post to the mailing list to raise awareness to the FAQ/Tutorial.