pyusb/pyusb #219: Before running the code, I need to pull and plug the usb motor once to work

woshiyushengzhou, 2018-11-19:
dear all:
import usb.core
import usb.util
import array

data_list = [0x01,0x01,0x08,0x00,0x00,0x00,0xc1,0x00,0x20,0x23,0xe8,0x03,0x00,0x00,0x03,0x00,0x00,0x00,
        0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
        0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
        0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
my_data = array.array('B',data_list)
# find our device
dev = usb.core.find(idVendor=1155, idProduct=22352)

# was it found?
if dev is None:
    raise ValueError('Device not found')

#solve Errno 16
reattach = False
if dev.is_kernel_driver_active(0):
    reattach = True
    dev.detach_kernel_driver(0)

print(dev.iManufacturer)
# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()
print("-----------------------------------------")

# get an endpoint instance
cfg = dev.get_active_configuration()
intf = cfg[(0,0)]
ep = usb.util.find_descriptor(
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)

assert ep is not None

# write the data
ep.write(my_data)
print("has write data...")

if reattach:
    dev.attach_kernel_driver(0)

The above is my code, I really need your help

mcuee, 2019-01-17:
Not a pyusb issue. You need to understand how the device works before using pyusb.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee