pyusb/pyusb #220: claim_interface methord hang up after usb device plugin-out 32 times.

leileigong, 2018-11-25:


leileigong, 2018-11-25:
i operate pyusb like this in one operation:
1)open device
2)write a simple data to the endpoint
3)read response data from the endpoint

after get response, i plugout my usb device.Then i plugin my usb device again and write and read as before.
but when i pluginout 32 times, just at the 33th operation, the write hang up at claim_interface
`    @methodtrace(_logger)
    def claim_interface(self, dev_handle, intf):
        _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
`

mcuee, 2019-01-17:
Make sure your device is still present in the system. What is the OS? Not a pyusb issue I will say.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee