pyusb/pyusb #221: USBError: [Errno 19] No such device (it may have been disconnected)

leileigong, 2018-11-25:
`    def OpenDevice(self):
        cfg = self.device.get_active_configuration()
        if cfg is None:
            # set the active configuration. With no arguments, the first
            # configuration will be the active one
            self.device.set_configuration()

        intf = cfg[(0, 0)]

        for i in range(intf.bNumEndpoints):
            e = intf[i]
            if usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT:
                self.epout = e
            else:
                if usb.util.endpoint_type(e.bmAttributes) == usb.util.ENDPOINT_TYPE_INTR:
                    self.epintr = e
                else:
                    self.epin = e
        return True` 

File "E:\ProductionTools\module\device\device.py", line 33, in OpenDevice
    cfg = self.device.get_active_configuration()
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\core.py", line 875, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\core.py", line 236, in get_active_configuration
    self.managed_open()
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\backend\libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\lei\.virtualenvs\ProductionTools-C2wJZE3m\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 19] No such device (it may have been disconnected)

mcuee, 2019-01-17:
Make sure your device is connected and appear properly in the Device Manager. And also make sure you have the right device driver like WinUSB installed to your device.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee

leileigong, 2020-01-17:
I have two task.
the first on detect the plugin an plugout of pyusb device.
the second one power on and power off the pyusb device.
when the device is pluged in. the first task will create another thread to operate the device.

my operation is 
1.power on
2.power off
3.reboot device.

`class CCIDDevice(object):
    def __init__(self, dev):
        self.dev = dev

    def power_on(self):
        # set the active configuration. With no arguments, the first
        # configuration will be the active one
        self.dev.set_configuration()

        # get an endpoint instance
        cfg = self.dev.get_active_configuration()
        intf = cfg[(0, 0)]

        self.ep_out = usb.util.find_descriptor(
            intf,
            # match the first OUT endpoint
            custom_match=lambda e: \
                (usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT) and \
                (usb.util.endpoint_type(e.bmAttributes) == usb.util.ENDPOINT_TYPE_BULK)
            )
        assert self.ep_out is not None

        self.ep_in = usb.util.find_descriptor(
            intf,
            # match the first OUT endpoint
            custom_match=lambda e: \
                (usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_IN) and \
                (usb.util.endpoint_type(e.bmAttributes) == usb.util.ENDPOINT_TYPE_BULK)
            )
        assert self.ep_in is not None

        self.seq = 1

    def power_off(self):
        usb.util.dispose_resources(self.dev)

    def transmit_apdu(self, apdu):
        message = b'\x6F' + len(apdu).to_bytes(4, byteorder='little') + b'\x00' + \
                  self.seq.to_bytes(1, byteorder='little') + b'\x00\x00\x00' + apdu
        # logging.debug(binascii.b2a_hex(message))
        self.ep_out.write(message)
        message = b''
        while True:
            packet = self.ep_in.read(self.ep_in.wMaxPacketSize, 12000)
            message += packet
            if len(packet) == self.ep_in.wMaxPacketSize:
                continue
            break
        # logging.debug(binascii.b2a_hex(message))
        self.seq += 1
        self.seq &= 0xFF

        return bytes(message[10:])`

but after run many times , i will got an error:
Traceback (most recent call last):
  File "E:\FNTools\apps\FN70P\FN70PFWDL\DeviceDownWidget.py", line 632, in upgrade_firm
    d.power_on()
  File "E:\FNTools\fntkdevice\fntkdevice.py", line 70, in power_on
    self.dev.power_on()
  File "E:\FNTools\fntkdevice\usbccid.py", line 15, in power_on
    self.dev.set_configuration()
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 147, in managed_set_configuration
    self.managed_open()
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\lei\.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

jonasmalacofilho, 2020-01-18:
Note: @leileigong has since then created a new issue for this in #292.