pyusb/pyusb #224: [op_set_configuration] failed, error -1 errno 32 when using set_configuration()

apeppels, 2018-12-12:
I'm getting an `Other error` (actually `errno 32`) when using `set_configuration()`:
 
```>>> import usb.core
>>> dev = usb.core.find(idVendor=0xtest, idProduct=0xtest)
>>> dev.bNumConfigurations
2
>>> dev.set_configuration(0x2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
```

With debug options:

```[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000001] [00003733] libusb: debug [libusb_init] created default context
[ 0.000014] [00003733] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.000027] [00003733] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000034] [00003733] libusb: debug [op_init] bulk continuation flag supported
[ 0.000037] [00003733] libusb: debug [op_init] zero length packet flag supported
[ 0.000045] [00003733] libusb: debug [op_init] sysfs can relate devices
[ 0.000049] [00003733] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000280] [00003735] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.000949] [00003733] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.000955] [00003733] libusb: debug [linux_get_device_address] scan usb1
[ 0.000977] [00003733] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.000980] [00003733] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.000984] [00003733] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001042] [00003733] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.001045] [00003733] libusb: debug [linux_get_device_address] scan 1-1
[ 0.001063] [00003733] libusb: debug [linux_get_device_address] bus=1 dev=113
[ 0.001064] [00003733] libusb: debug [linux_enumerate_device] busnum 1 devaddr 113 session_id 369
[ 0.001066] [00003733] libusb: debug [linux_enumerate_device] allocating new device for 1/113 (session 369)
[ 0.001082] [00003733] libusb: debug [linux_get_parent_info] Dev 0x555d2aa28ea0 (1-1) has parent 0x555d2aa30570 (usb1) port 1
[ 0.001119] [00003733] libusb: debug [linux_get_device_address] getting address for device: 1-7 detached: 0
[ 0.001121] [00003733] libusb: debug [linux_get_device_address] scan 1-7
[ 0.001136] [00003733] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.001140] [00003733] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.001143] [00003733] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.001163] [00003733] libusb: debug [linux_get_parent_info] Dev 0x555d2aa309e0 (1-7) has parent 0x555d2aa30570 (usb1) port 7
[ 0.001202] [00003733] libusb: debug [linux_get_device_address] getting address for device: 1-9 detached: 0
[ 0.001205] [00003733] libusb: debug [linux_get_device_address] scan 1-9
[ 0.001228] [00003733] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.001231] [00003733] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.001233] [00003733] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.001254] [00003733] libusb: debug [linux_get_parent_info] Dev 0x555d2aa30630 (1-9) has parent 0x555d2aa30570 (usb1) port 9
[ 0.001291] [00003733] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001295] [00003733] libusb: debug [linux_get_device_address] scan usb2
[ 0.001316] [00003733] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001319] [00003733] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001322] [00003733] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001345] [00003733] libusb: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.001352] [00003733] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.001354] [00003733] libusb: debug [usbi_add_pollfd] add fd 9 events 1
[ 0.001413] [00003733] libusb: debug [libusb_get_device_list] 
[ 0.001465] [00003733] libusb: debug [libusb_get_device_descriptor] 
[ 0.001544] [00003733] libusb: debug [libusb_get_device_descriptor] 
[ 0.001617] [00003733] libusb: debug [libusb_get_device_descriptor] 
[ 0.001662] [00003733] libusb: debug [libusb_get_device_descriptor] 
[ 0.001722] [00003733] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.001728] [00003733] libusb: debug [parse_endpoint] skipping descriptor 24
[ 0.001763] [00003733] libusb: debug [libusb_get_config_descriptor] index 1
[ 0.001768] [00003733] libusb: debug [parse_endpoint] skipping descriptor 24
[ 0.001770] [00003733] libusb: debug [parse_endpoint] skipping descriptor b
[ 0.001801] [00003733] libusb: debug [libusb_open] open 1.113
[ 0.001815] [00003733] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.001824] [00003733] libusb: debug [libusb_set_configuration] configuration 2
[ 0.002520] [00003733] libusb: error [op_set_configuration] failed, error -1 errno 32
Traceback (most recent call last):
  File "script.py", line 8, in <module>
    dev.set_configuration(0x2)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
[ 0.004915] [00003733] libusb: debug [libusb_close] 
[ 0.004931] [00003733] libusb: debug [usbi_remove_pollfd] remove fd 10
[ 0.004954] [00003733] libusb: debug [libusb_exit] 
[ 0.004957] [00003733] libusb: debug [libusb_exit] destroying default context
[ 0.004963] [00003733] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.004966] [00003733] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.004973] [00003733] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 0.004977] [00003733] libusb: debug [handle_events] poll() returned 0
[ 0.004980] [00003733] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.004983] [00003733] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.004986] [00003733] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.004989] [00003733] libusb: debug [libusb_unref_device] destroy device 1.113
[ 0.004991] [00003733] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.004994] [00003733] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.005004] [00003733] libusb: debug [usbi_remove_pollfd] remove fd 9
[ 0.005021] [00003735] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

mcuee, 2019-01-17:
What is the OS? What are the two valid configurations?

apeppels, 2019-01-18:
I'm running Debian 9, the problem seems to be that I need to reset the USB before switching (this also fails the first time, I need to do it twice)

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee