pyusb/pyusb #226: Threading issue using pyusb

yohan-m, 2018-12-20:
Hello,

I'm currently having an issue writing a multithreaded app with pyusb. Here is a quick description of my application:
- 1 discovery thread that periodically (every second) check for connected devices (can be deactivated and I did so to figure out the bug)
- 1 thread per device, periodically (every 10ms) it writes to device if some data is cached for sending, and then read once and store received frame if there is one. Read and write actions are executed if no discovery is ongoing (I checked that there are no issues between discovery and device thread for IO access)
- the main thread that receives write requests (forward them to device's thread queue) and retrieves stored received frames to print them

While using 1 device, there is no issue.
If I start connecting 2 devices, both will hang on their read instruction (interrupt read for HID devices) at some point. I pinpointed the issue to be a simultaneous read request (or at least concurrent ones while a device is waiting for its read timeout).

Thing is, I'm using a 1ms timeout (had to increase it to 10s to force the error to happen sooner), and I thought using unblocking IO would be the solution but it seems like it does not exist on read/write instructions (timeout=None uses the default one, timeout=0 is infinite blocking, a positive value is a millisecond timeout and negative values are not allowed if I'm not mistaken).

So question is: is there a way to have non blocking IO calls that return immediately (and thus would not be able to be concurrent ones) or is there a version of pyusb that supports multithreading?

Note that for now, I went back to doing all actions in the main thread but that's not statisfying to me if there happens to be other blocking issues on IO..

Regards,

Yohan

mcuee, 2019-11-29:
Not an expert here. Comments are welcome. Enhancement are also welcome.

jonasmalacofilho, 2019-12-13:
@yohan-m,

What backend were you using?  Can you show me some sample code?

You might also want to take a look at _[libusb: Multi-threaded applications and asynchronous I/O](http://libusb.sourceforge.net/api-1.0/mtasync.html)._

jonasmalacofilho, 2020-01-05:
@yohan-m I would like to take a look at this, but I do need so sample code and additional details like OS, backend, library versions...

Btw I'm sorry it took us a while to get to this issue but, as you've probably noticed by now, the project had a brief period of hibernation.  But I'm the new principal maintainer, and I'll be happy to work with you on this.

closed 2020-02-16 by jonasmalacofilho

mcuee, 2021-08-02:
Looks like #63 and #64 are both needed in this case to improve the performance.