pyusb/pyusb #230: Not able to read data from USB:Pyusb Windows error:USBError: [Errno None] libusb0-dll:err [claim_interface] could not claim interface 1, win error: The requested resource is in use

dhingrachirag, 2019-02-18:
Hi 

I am trying to read data from USB using pyusb in python.I am able to get all the configuration of devices but when i am trying to read data .I am getting the below error:

USBError: [Errno None] libusb0-dll:err [claim_interface] could not claim interface 1, win error: The requested resource is in use.

PF the code also what i had written:

import usb.core
test = usb.core.find(idVendor=0x0ghe, idProduct=0x0241)
print test

test.set_configuration()

for i in range(0, 20):
    while True:
        try:
            test = test.read(0x81, 8, timeout=50)
            break
        except usb.core.USBError, e:            
            if str(e).find("timeout") >= 0:
                pass
            else:
                raise IOError("USB Error: %s"%str(e))

    print test

Below are my questions:

   1) How to read data from USB in host machine every second whenever we are doing any operation in usb connected device ?
   2) Why this error is coming while reading data from endpoints ?
   3) What's the efficient way to read inputs from USB using pyusb for any button press whatever we are doing in device ?

p144407, 2019-03-17:
I have similar issue as well.

I have several control transfers that work correctly

-->  wrConfirm       = self.device.ctrl_transfer(bmRequest, 
                                           bRequest, 
                                           wValue, 
                                           wIndex, 
                                           buferOut)

I do the following
-->device.read(0x81,512,5000)
which returns:
-->USBError: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, win error: The requested resource is in use.\r\n'

I read somewhere, I should try to detach kernel driver?

But this: 
-->device.detach_kernel_driver(interface)
returns this:
-->function 'usb_detach_kernel_driver_np' not found

Full traceback for bulk read starts here:

Traceback (most recent call last):

  File "<ipython-input-45-caa7b40ff5a2>", line 1, in <module>
    usb.util.claim_interface(chandra.device, 0)

  File "C:\Users\jerome.crocco\AppData\Local\Continuum\anaconda3\envs\dpixDev\lib\site-packages\pyusb-1.0.2-py3.5.egg\usb\util.py", line 192, in claim_interface
    device._ctx.managed_claim_interface(device, interface)

  File "C:\Users\jerome.crocco\AppData\Local\Continuum\anaconda3\envs\dpixDev\lib\site-packages\pyusb-1.0.2-py3.5.egg\usb\core.py", line 89, in wrapper
    return f(self, *args, **kwargs)

  File "C:\Users\jerome.crocco\AppData\Local\Continuum\anaconda3\envs\dpixDev\lib\site-packages\pyusb-1.0.2-py3.5.egg\usb\core.py", line 154, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)

  File "C:\Users\jerome.crocco\AppData\Local\Continuum\anaconda3\envs\dpixDev\lib\site-packages\pyusb-1.0.2-py3.5.egg\usb\backend\libusb0.py", line 508, in claim_interface
    _check(_lib.usb_claim_interface(dev_handle, intf))

  File "C:\Users\jerome.crocco\AppData\Local\Continuum\anaconda3\envs\dpixDev\lib\site-packages\pyusb-1.0.2-py3.5.egg\usb\backend\libusb0.py", line 418, in _check
    raise USBError(errmsg, ret)

USBError: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, win error: The requested resource is in use.\r\n'

xobs, 2019-04-20:
Windows doesn't allow you to connect to USB HID devices, as that could be used to implement a keylogger.

What you _can_ do is connect to those HID devices and send control packets back and forth.  This can be used, for example, to do a DFU update on a keyboard.  Earlier versions of libusb would fail when they tried to open a USB HID device, because Windows won't actually let you fully bind to the device.  I don't know if that ever got fixed.

mcuee, 2019-11-29:
For HID device under Windows, you should try to use pywinusb.

mcuee, 2019-11-29:
And if you want to use libusb-win32 backend, you need to replace the HID driver with libusb0.sys.

As for libusb-1.0 backend, it does support HID but there are limitations. You can try it. If it does not work, try to replace the driver with WinUSB.

mcuee, 2019-11-29:
For further support, please use pyusb mailing list.

closed 2019-11-29 by mcuee

shuvragood, 2022-10-24:
Please do compare both. Might be the issue can be identified with USB Time out error .
I am having all code with me (https://drive.google.com/drive/folders/1hKyy2O2HLyh9o4wxKsPFoyXOMyaorkv7)

https://www.duo.uio.no/bitstream/handle/10852/82665/5/Olav-Bjerke---masteroppgave.pdf .

Getting USB Time out error while replicating this into CY8CKIT-059