pyusb/pyusb #231: Out-of-date manifest

turboencabulator, 2019-03-09:
Two warnings during the build:
```
reading manifest file 'pyusb.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'ChangeLog'
warning: no files found matching 'ReleaseNotes.rst'
writing manifest file 'pyusb.egg-info/SOURCES.txt'
```
ReleaseNotes.rst went away in 1d54b1a4c58a5c852d66b2374fb1d40456f699f3.  I can't find any instance of ChangeLog.

mcuee, 2019-11-29:
Patches in the form of Pull Request are welcome. Thanks.

jonasmalacofilho, 2019-11-29:
ChangeLog is generated by `gencl.sh` during packaging with `deploy.sh`.

mcuee, 2019-11-30:
In this case, this is not really a bug. So we can close it.

jonasmalacofilho, 2019-11-30:
ReleaseNotes.rst is still wrong though.

mcuee, 2019-12-07:
Yes you are right that ReleaseNotes.rst is missing.

`C:\work\pyusb\pyusb>c:\Python27\python.exe setup.py sdist
running sdist
running egg_info
writing pyusb.egg-info\PKG-INFO
writing top-level names to pyusb.egg-info\top_level.txt
writing dependency_links to pyusb.egg-info\dependency_links.txt
reading manifest file 'pyusb.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'ReleaseNotes.rst'
writing manifest file 'pyusb.egg-info\SOURCES.txt'
running check
creating pyusb-1.0.2
creating pyusb-1.0.2\docs
creating pyusb-1.0.2\pyusb.egg-info
creating pyusb-1.0.2\usb
creating pyusb-1.0.2\usb\backend
copying files to pyusb-1.0.2...
copying ACKNOWLEDGEMENTS -> pyusb-1.0.2
copying ChangeLog -> pyusb-1.0.2
copying LICENSE -> pyusb-1.0.2
copying MANIFEST.in -> pyusb-1.0.2
copying README.rst -> pyusb-1.0.2
copying setup.cfg -> pyusb-1.0.2
copying setup.py -> pyusb-1.0.2
copying docs\faq.rst -> pyusb-1.0.2\docs
copying docs\tutorial.rst -> pyusb-1.0.2\docs
copying pyusb.egg-info\PKG-INFO -> pyusb-1.0.2\pyusb.egg-info
copying pyusb.egg-info\SOURCES.txt -> pyusb-1.0.2\pyusb.egg-info
copying pyusb.egg-info\dependency_links.txt -> pyusb-1.0.2\pyusb.egg-info
copying pyusb.egg-info\top_level.txt -> pyusb-1.0.2\pyusb.egg-info
copying usb\__init__.py -> pyusb-1.0.2\usb
copying usb\_debug.py -> pyusb-1.0.2\usb
copying usb\_interop.py -> pyusb-1.0.2\usb
copying usb\_lookup.py -> pyusb-1.0.2\usb
copying usb\_objfinalizer.py -> pyusb-1.0.2\usb
copying usb\control.py -> pyusb-1.0.2\usb
copying usb\core.py -> pyusb-1.0.2\usb
copying usb\legacy.py -> pyusb-1.0.2\usb
copying usb\libloader.py -> pyusb-1.0.2\usb
copying usb\util.py -> pyusb-1.0.2\usb
copying usb\backend\__init__.py -> pyusb-1.0.2\usb\backend
copying usb\backend\libusb0.py -> pyusb-1.0.2\usb\backend
copying usb\backend\libusb1.py -> pyusb-1.0.2\usb\backend
copying usb\backend\openusb.py -> pyusb-1.0.2\usb\backend
Writing pyusb-1.0.2\setup.cfg
Creating tar archive
removing 'pyusb-1.0.2' (and everything under it)`

commit 1559ff23 (2019-12-10): Remove ReleaseNotes.txt from the manifest

ReleaseNotes.rst had long been superseded by a git-log changelog, and
was finally removed in 1d54b1a4c58a ("Release notes not necessary
anymore").

The changelog is generated by gencl.sh during packaging (see: deploy.sh).

Fixes: #231 ("Out-of-date manifest")
Reported-by: @turboencabulator
closed 2019-12-10 by jonasmalacofilho