pyusb/pyusb #232: Segfault if libusb fails to initialize

eric-wieser, 2019-03-11:
Caused by the following
```python
class _LibUSB(usb.backend.IBackend):
    @methodtrace(_logger)
    def __init__(self, lib):
        usb.backend.IBackend.__init__(self)
        self.lib = lib
        self.ctx = c_void_p()
        _check(self.lib.libusb_init(byref(self.ctx)))

    @methodtrace(_logger)
    def _finalize_object(self):
        self.lib.libusb_exit(self.ctx)
```

If `libusb_init` returns an error, then `self.ctx` is `NULL`, and `libusb_exit(NULL)` segfaults.

tedsta, 2019-05-24:
I think I'm also running into this issue.

Relevant portion of `faulthandler` output:
```
Fatal Python error: Segmentation fault
 Current thread 0x730ff470 (most recent call first):
   File "/usr/local/lib/python3.5/dist-packages/usb/backend/libusb1.py", line 684 in __del__
   File "/usr/local/lib/python3.5/dist-packages/usb/backend/libusb1.py", line 901 in get_backend
   File "/usr/local/lib/python3.5/dist-packages/usb/core.py", line 859 in find
```

jonasmalacofilho, 2019-11-29:
LGTM.

@eric-wieser can you submit a patch?

jonasmalacofilho, 2019-11-29:
Fixed with #279 ("Don't call libusb_exit on a null pointer").

Thank you!

closed 2019-11-29 by jonasmalacofilho