pyusb/pyusb #233: usb.core.find( find_all=True, ... ) should return None when no match is found.

sunbearc22, 2019-03-27:
I noticed the returned value between this two lines of code  is different when no device was connect.
```
devices = usb.core.find( idVendor=Vendor, iProduct = Product, )
devices = usb.core.find( find_all=True, idVendor=Vendor, iProduct = Product, )
```
For the 1st case,  devices returned `None`. 
For the 2nd case,  devices returned  `<generator object device_iter at 0x7fd3fac0b048>`

My expectation is that `None` should be returned for both cases.

mcuee, 2019-11-29:
Patches welcome.

jonasmalacofilho, 2019-11-29:
I disagree.

In the first case the caller expects a binary result (zero or one device), but in the second an iterable is needed (as zero, one or more devices are possible).

An empty iterable is IMO the more semantically correct result for when there are no matching devices and `find_all=True`.  It is also more practical, as otherwise the caller would always need to check the response against `None`.

Additionally, this would be a breaking change.

---

TL; DR: `find_all=False` and `find_all=True` have different uses, and the latter presumes iteration.

mcuee, 2019-11-30:
In this case, I think this is not a bug and we can close this.

closed 2019-11-30 by mcuee