pyusb/pyusb #237: USBError: [Errno 13] Access denied (insufficient permissions) Ubuntu 16.04

nikhil4595, 2019-05-20:
Hi ,

On running this particular _pycrafter4500_ module in my python terminal, I kept getting the following error. Any ideas, what might be the issue here ?

`In[4]: import pycrafter4500
In[5]: pycrafter4500.power_up()
Traceback (most recent call last):
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/IPython/core/interactiveshell.py", line 3296, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-5-08c5317d7bd7>", line 1, in <module>
    pycrafter4500.power_up()
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/pycrafter4500/pycrafter4500.py", line 524, in power_up
    with connect_usb() as lcr:
  File "/usr/lib/python3.5/contextlib.py", line 59, in __enter__
    return next(self.gen)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/pycrafter4500/pycrafter4500.py", line 82, in connect_usb
    device.set_configuration()
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/core.py", line 147, in managed_set_configuration
    self.managed_open()
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/nikhil/Complete pycharm project/untitled2/venv2/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
`
Thanks in advance

Regards,
Nikhil

ixjlyons, 2019-05-20:
You probably just need to install a udev rule. I don't know about your device, but here's a nice generic writeup on writing them http://www.reactivated.net/writing_udev_rules.html

Something like this should work for USB devices (substitute your product and vendor IDs):

```
SUBSYSTEM=="usb", ATTR{idVendor}=="ffff", ATTR{idProduct}=="ffff", MODE="0666"
```

Log out then back in after adding the rule to `/etc/udev/rules.d/` and your account should be able to access the device.

nikhil4595, 2019-05-21:
Hi Kenneth,

I have actually tried this before, that is why I am confused,as to why I am still getting this error. Because almost every solution to this problum is to establish a new udev rule for the device.
This is the solution from one of the users in the TI forum

[image]

ixjlyons, 2019-05-21:
Does the script/program work if you run it as root? If so, I'm pretty sure you just need to make sure your udev rule correctly matches your device and gives sufficient read and write permissions. Maybe try using `SUBSYSTEMS` and `ATTRS` (note the appended "S") in the rule I posted above. The `udevadm` tool should be helpful in debugging.

If it doesn't even work when run as root, I'm not sure what the problem could be. The error ultimately comes from libusb, so be sure not to limit your searches for a solution to pyusb. Good luck and sorry I can't really offer much more help.

mcuee, 2019-11-29:
Please use mailing list for technical support.

closed 2019-11-29 by mcuee