pyusb/pyusb #239: usb.read() always returns the same

vstadnytskyi, 2019-06-06:
I am using USB library in python. I can see the device, i can configure it. but when I read out of it I always get the same stuff `array('B', [17, 96])`. (see full code below)

the lsusb command shows it is a FTDI USB device
` Bus 009 Device 008: ID 0403:faf0 Future Technology Devices International, Ltd `

in past, i was able to communicate with other usb devices? I don't understand what is happening with this one. Could someone point to the right direction? I looked into pylibftdi but with that library, I cannot even see this usb device.

```
>>> import usb
serial_number = '83836244'
dev = None
        devices = list(usb.core.find(idVendor=0x0403, idProduct=0xFAF0, find_all = True))
        for dev in devices:
            if dev.serial_number == serial_number:
                break
       dev = dev

reattach = False
if self.dev.is_kernel_driver_active(0):
    reattach = True
    self.dev.detach_kernel_driver(0)
# set the active configuration. With no arguments, the first
# configuration will be the active one
self.dev.set_configuration()
# get an endpoint instance
cfg = self.dev.get_active_configuration()
intf = cfg[(1,1)]
epo = usb.util.find_descriptor(
                              intf,
                              # match the first OUT endpoint
                              custom_match = \
                              lambda e: \
                              usb.util.endpoint_direction(e.bEndpointAddress) == \
                              usb.util.ENDPOINT_OUT)

epi = usb.util.find_descriptor(
                              intf,
                              # match the first IN endpoint
                              custom_match = \
                              lambda e: \
                              usb.util.endpoint_direction(e.bEndpointAddress) == \
                              usb.util.ENDPOINT_IN)

assert self.epo is not None
assert self.epi is not None
epi.wMaxPacketSize = 72000
epo.wMaxPacketSize = 72000
epi.bmAttributes = 1
epi.bInterval = 100
usb_buff = int(self.epi.wMaxPacketSize/100)

dev.read(self.epi,100,1000)

array('B', [17, 96])
```

vstadnytskyi, 2019-06-07:
I have also published this question on [StackOverflow](https://stackoverflow.com/questions/56486431/pyusb-read-always-returns-the-same-value)

ixjlyons, 2019-06-07:
My first thought: do you know the protocol of the device you're trying to interact with? Perhaps you have send a control transfer with a certain payload to get something useful out, for example?

vstadnytskyi, 2019-06-07:
@ixjlyons I know the protocol and I was able to communicate with it on Windows after switching this USB device to VCP (Virtual COM port). By doing so I am losing information such as serial numbers that are stored on USB chip and are not accessible when the device is in VCP mode. I also want to switch from Windows to Linux.

ixjlyons, 2019-06-07:
Hmm, possibly this SO post could point in the right direction: https://stackoverflow.com/q/33649296

mcuee, 2019-11-29:
Please use mailing list for technical support.

mcuee, 2019-11-29:
And it is better either to use pyftdi (based on pyusb) or libftdi python bindings for ftdi device. 

By using pyusb, you need to lower level protocol of the device.

closed 2019-11-29 by mcuee