pyusb/pyusb #240: VISA USB, GPIB

Marcel191536, 2019-06-27:
Hi, I am trying this:
   import visa
   rm = visa.ResourceManager()
   rm.list_resources()
but the result is this:
   ()
When I write 
   !python -m visa info
result looks like this:
   Machine Details:
      Platform ID:    Windows-10-10.0.17134-SP0
      Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

   Python:
      Implementation: CPython
      Executable:     C:\Users\HP\Anaconda3\python.exe
      Version:        3.7.3
      Compiler:       MSC v.1915 64 bit (AMD64)
      Bits:           64bit
      Build:          Mar 27 2019 17:13:21 (#default)
      Unicode:        UCS4

   PyVISA Version: 1.9.1

   Backends:
      ni:
         Version: 1.9.1 (bundled with PyVISA)
         #1: C:\windows\system32\visa32.dll:
            found by: auto
            bitness: 64
            Vendor: National Instruments
            Impl. Version: 17827072
            Spec. Version: 5244928
         #2: C:\windows\system32\visa64.dll:
            found by: auto
            bitness: 64
            Vendor: National Instruments
            Impl. Version: 17827072
            Spec. Version: 5244928
      py:
         Version: 0.3.1
         ASRL INSTR: Available via PySerial (3.4)
         USB INSTR:
            **PyUSB does not seem to be properly installed.
            Please refer to PyUSB documentation and 
            install a suitable backend like 
            libusb 0.1, libusb 1.0, libusbx, 
            libusb-win32 or OpenUSB.
            No backend available**
         USB RAW:
            **PyUSB does not seem to be properly installed.
            Please refer to PyUSB documentation and 
            install a suitable backend like 
            libusb 0.1, libusb 1.0, libusbx, 
            libusb-win32 or OpenUSB.
            No backend available**
         TCPIP INSTR: Available 
         TCPIP SOCKET: Available 
         GPIB INSTR:
            **Please install linux-gpib to use this resource type.
            No module named 'gpib'**
QUESTIONS:
1. ---why do I get: PyUSB does not seem to be properly installed
    ---even if I tried: !pip install pyusb
2. ---it is said: PyUSB comes with builtin backends for libusb 0.1, libusb 1.0 and OpenUSB. You can write your own backend if you desire to.
   ---why do have to install a suitable backend like libusb 0.1,...? 
3. ---how to install those (libusb 0.1,...) backends on Windows?
4. ---how to install linux-gpib?
5. ---I WANT TO GET: GPIB INSTR: Available
                                   USB INSTR: Available
                                   USB Raw: Available
Any (easy to understand) advice is welcome!

ixjlyons, 2019-06-27:
The statement that pyusb has a variety of backends built in is somewhat misleading. Actually, pyusb contains code to interface with a variety of backend implementations, providing a unified high-level interface to them -- the pyusb API. `pip install pyusb` will *not* automatically choose and install a backend for you. There is a note in the README about installing libusb on Windows, though keep in mind it may be outdated: https://github.com/pyusb/pyusb#installing

For GPIB, I'm not sure. I would recommend asking the PyVISA team.

mcuee, 2019-11-29:
Please use mailing list for technical support.

closed 2019-11-29 by mcuee