pyusb/pyusb #241: linux-gpib

GitHub191536, 2019-06-27:
Can anyone tell me please, how to install linux-gpib on Windows? (Please install linux-gpib to use this resource type. No module named 'gpib')
Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\HP\Anaconda3\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1915 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 27 2019 17:13:21 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

wolf99, 2019-08-15:
Seems like some folks may have got it working here: https://github.com/pyvisa/pyvisa-py/issues/105

mcuee, 2019-11-29:
Not really related to pyusb.

closed 2019-11-29 by mcuee