pyusb/pyusb #242: usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_setup_async] invalid endpoint 0x82'

spasimir21, 2019-07-05:
I`m sure that the endpoint exists because I can print info about it.
I use clear_hault() before write.
But still this error message pops out.
This endpoint is the only one on the single configuration for this device.
I`m sure that there is only one configuration.
the device is an HID device but I only installed libusb-win32 using zadig on one configuration(the second one (1) )
the device is a keyboard and has two interfaces:
0 - keyboard stuff
1 - rgb stuff

Here is the full error:

Traceback (most recent call last):
  File "usbstuff.py", line 14, in <module>
    e.write(b"\x00")
  File "D:\anaconda\lib\site-packages\usb\core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "D:\anaconda\lib\site-packages\usb\core.py", line 948, in write
    self.__get_timeout(timeout)
  File "D:\anaconda\lib\site-packages\usb\backend\libusb0.py", line 551, in intr_write
    timeout)
  File "D:\anaconda\lib\site-packages\usb\backend\libusb0.py", line 616, in __write
    timeout
  File "D:\anaconda\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_setup_async] invalid endpoint 0x82'

ixjlyons, 2019-07-15:
Looks like you're trying to write to an input endpoint (in/out is from the perspective of the host). I'm not all that familiar with USB HID, but I would imagine you have to use control transfers to send information to the device (endpoint 0 is always present).

mcuee, 2019-11-29:
Please use mailing list for technical support. Github is for issue reporting and pull-request.

closed 2019-11-29 by mcuee