pyusb/pyusb #244: GitHub thinks this is Apache 2.0?

jakirkham, 2019-08-05:
Not sure why this is or how to fix it ATM, but it appears GitHub thinks this is Apache 2.0 despite being BSD-3-Clause. Perhaps worth raising to GitHub to see what's up with their license identification tool.

[image]

jonasmalacofilho, 2019-09-18:
Apparently the license changed with #201 ("Moving to the standard Apache 2.0 License").

Sure, there were oddities... but was it really necessary to change the to Apache instead of just restating that everything was under the BSD license?

Moving forward this could cause some issues with projects under the GPL v2.¹  I guess we'll have to see...

¹ [Apache License v2.0 and GPL Compatibility](https://www.apache.org/licenses/GPL-compatibility.html)

jonasmalacofilho, 2019-11-29:
I think we should _consider_ reverting back to a BSD 3-clause license (with fixed file notices).  Obviously we would need to get buy-in for the few contributions since the change, but it's IMO better than pushing a more restrictive license downstream out-of-the-blue.

That is, unless all of the maintainers feel that this is important.  I for one do not, as we _all_ inherited this codebase from Wander (and I wouldn't have personally chosen Apache).  But I'm also not opposed to it, if done right.  Still, in this case I would like to see the major version number bumped.

mcuee, 2019-11-30:
I agree to revert back to the BSD 3-clause license.
@SimplicityGuy Any objection?

SimplicityGuy, 2019-12-07:
I don't have objections to doing so.

PR #281 (2019-12-07): back to bsd 3 clause
Fixes #244
closed 2019-12-07 by SimplicityGuy