pyusb/pyusb #286: reading fron an endpoint produces tb

kdschlosser, 2019-12-29:
Stackless Python 2.7.16 (CPython)
Windows 7 x64 SP1
libusb-1.0

I am trying to read from a bulk endpoint and I get the following traceback..

```python
Traceback (most recent call last):
  File "\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "\pyWinMCERemote\pyWinMCERemote\ioctl.py", line 1624, in __receive_loop
    print endpoint.read(packet_size)
  File "\lib\site-packages\usb\core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "\lib\site-packages\usb\core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "\lib\site-packages\usb\core.py", line 215, in setup_request
    intf, ep = self.get_interface_and_endpoint(device, endpoint_address)
  File "\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "\lib\site-packages\usb\core.py", line 225, in get_interface_and_endpoint
    for intf in self.get_active_configuration(device):
  File "\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "\lib\site-packages\usb\core.py", line 236, in get_active_configuration
    self.managed_open()
  File "\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "\lib\site-packages\usb\core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "\lib\site-packages\usb\backend\libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "\lib\site-packages\usb\backend\libusb1.py", line 593, in _check
    raise NotImplementedError(_strerror(ret))
NotImplementedError: Operation not supported or unimplemented on this platform
```

is reading from a USB device not supported under windows??

jonasmalacofilho, 2019-12-29:
The issue has to do with libusb not being able to _open_ the device and returning `LIBUSB_ERROR_NOT_SUPPORTED`.

I need more details to pinpoint why, but one possible cause is that your device is *not* a HID nor is there a libusb-compatible driver installed for it.

If you suspect this to be the case, try installing a suitable driver with Zadig; without knowing more about the device I suggest you pick WinUSB.  See [How to use libusb under Windows](https://github.com/libusb/libusb/wiki/FAQ#How_to_use_libusb_under_Windows) for more information.

kdschlosser, 2019-12-29:
I am not 100% sure what category this device falls into. 
ok well these are the files the device is using.

CIRCoInst.dll
hhdusbh64.sys
usbcir.sys

jonasmalacofilho, 2019-12-29:
It does appear that my initial suspicion was right: you need to switch to a libusb-compatible driver (more information in my previous comment).

---

@mcuee, do you known if usbdk would help here?  _(would obviously depend on #200)_

kdschlosser, 2019-12-30:
with my specific use case and the users that use my software I do not want to have them swap drivers around. I do not think that it would go over all that well and someone could easily make a mistake and end up blue screening their PC.. 

One would have thought that the ability to do something like this would be available through the use of DeviceIoControl I have not come across a way to do it other then using WinUSB or HIDAPI functions and those require the matching drivers to be installed.

Even tho the CIR device does fall into the HID category the HIDAPI functions do not work. neither do the WinUSB ones. I have also tried using the "HID" wrapper driver that gets installed along with the device driver and this one is not able to be accessed either.

jonasmalacofilho, 2019-12-30:
> with my specific use case and the users that use my software I do not want to have them swap drivers around. I do not think that it would go over all that well and someone could easily make a mistake and end up blue screening their PC..

Your software installer could do that for your users.  Or you could advise them to use Zadig, as I do with liquidctl.

> One would have thought that the ability to do something like this would be available through the use of DeviceIoControl I have not come across a way to do it other then using WinUSB or HIDAPI functions and those require the matching drivers to be installed.

HIDAPI (and libusb for that matter) should work well with the generic Microsoft HID Driver (assuming your device is indeed a *minimally compliant* HID).  Can't your device be used just with that default driver?

> Even tho the CIR device does fall into the HID category the HIDAPI functions do not work. neither do the WinUSB ones.

Well, this is again a driver issue.  You can't expect generic libraries to be able to talk to arbitrary vendor-supplied custom drivers.

In fact, I'm even surprised that your device is a HID.  Sure, I've seem some HIDs that require their own drivers, but it isn't that common.  And I'm not sure how Windows handles custom HID drivers (on Linux the custom driver can choose to block or allow the default hiddev/hidraw).

> I have also tried using the "HID" wrapper driver that gets installed along with the device driver and this one is not able to be accessed either.

I'm not sure what you mean by this HID wrapper.  Can you elaborate?

---

Can you post the vendor and product IDs of this device?

I would also like to see the USB descriptors (class, interfaces, endpoints, etc.), if it isn't too much work.

jonasmalacofilho, 2019-12-30:
You can also take a look at [UsbDk](https://github.com/daynix/UsbDk), as mentioned in the libusb FAQ linked before.

There are a couple things missing for it in PyUSB, but you can find a workaround in #200 ("How to use usbdk as backend [...]").

kdschlosser, 2019-12-30:
This is not a vendor supplied driver. It is a Windows ONLY device and a Microsoft provided driver. 

From what I have gotten from reading the specifications for the device that Microsoft makes the vendors conform to is that the device is a generic USB device. but once it gets plugged into the USB there is a OS descriptor the device is able to provide (Windows only) that Windows uses to determine what kind of a "special generic device" it is. from that descriptor Windows is able to install wrapper drivers. Now I know that we would not be able to do anything with these wrapper drivers are they  are not hardware. I want to read the data that is being passed to and from the hardware device without changing the behavior of the device. at some point when I have collected enough information on the data I might write a driver so the devices can be used on other platforms. There is no type for this device defined in the USB specification that is why Microsoft uses that Windows only descriptor, to extend the device types and also a bunch of other information as well.. I have used usbpcap but unfortunately it does not support those descriptors so I am not able to peek at what is being passed.
I did try another usb analyzer.. and the same thing. does not have support for reading those descriptors. It appears that the USB specification is what gets followed and not anything windows related even tho the analyzers are made only for use on Windows..

jonasmalacofilho, 2019-12-31:
I'm guessing from the driver and your repositories that we're talking about a Windows IR remote.  If that's the case, this is just as much a Windows-only device as any other mainstream product... I was using one of these on Linux back in 2014.

In fact, the [mceusb](https://github.com/torvalds/linux/blob/master/drivers/media/rc/mceusb.c) Linux driver could be a good resource for you, besides the [MS specification ](https://forum.team-mediaportal.com/download/2014/04/196064_Windows-Media-Center-RC-IR-Collection-Green-Button-Specification-03-08-2011-V2_1.pdf) (which is probably the document you were referring to).  But that driver is licensed under the GPL, so mind how you license any code derived from it.

To interact with the device with PyUSB/libusb on Windows, but without switching drivers, I really don't see any alternative other than to use UsbDk.

I'll close the issue, since this is a limitation of the device and currently in-use drivers.

closed 2019-12-31 by jonasmalacofilho

mcuee, 2020-01-03:
Yes I agree that the only possible option is to use usbdk driver.