pyusb/pyusb #288: OSError: exception: access violation writing 0x00000014 in libusb1.py

cyberkryption, 2020-01-05:
I am trying to detect if a particular usb stick is inserted on Windows 10 version 1903

I have python 3.65 installed

**Installed via pip**
**Current libraries installed**
libusb (1.0.22b8)  - Python binding for the libusb C library (ctypes/cffi-based libusb).   INSTALLED: 1.0.22b8 (latest)
pylibusb (0.2)     - libusb wrapper (ctypes based)
libusb1 (1.7.1)    - Pure-python wrapper for libusb-1.0   INSTALLED: 1.7.1 (latest)

Also have c bindings from libusb installed version 1.0.22 from libusb.info as shown below

copy MS64/dll/libusb*.dll c:\windows\system32
copy MS32/dll/libusb*.dll c:\windows\SysWOW64

If I try the following code when the usb stick is inserted

```
import sys
import usb.core
import usb.util

def check_usb_connect():
    # check specific usb device pluggedin
    dev = usb.core.find(idVendor=0x8644, idProduct=0x8003)
    if dev is None:
        print("Not Connected")
    else:
        print("Usb device idVendor=0x8644, idProduct=0x8003 Connected")
           
def main():
    #while dev is None:
    check_usb_connect()
    
if __name__ == '__main__':
    main()
```

I get the following exception

Usb device idVendor=0x8644, idProduct=0x8003 Connected
Exception ignored in: <bound method _AutoFinalizedObjectBase.__del__ of <usb.backend.libusb1._Device object at 0x039D5AB0>>
Traceback (most recent call last):
  File "C:\Users\XXX\security-tools\Python-3.65\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__self.finalize()
  File "C:\Users\XXX\security-tools\Python-3.65\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize    self._finalizer()
  File "C:\Users\XXX\security-tools\Python-3.65\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\XXX\security-tools\Python-3.65\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\XXX\security-tools\Python-3.65\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\XXX\security-tools\Python-3.65\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)

OSError: exception: access violation writing 0x00000014

The following code produces no error when the usb stick in inserted

```
import sys
import usb.core

dev = usb.core.find(idVendor=0x8644, idProduct=0x8003)

if dev is None:
    print("USB device is not connected")
else:
    print("Usb device idVendor=0x8644, idProduct=0x8003 Connected")
```

Can you help me with the above error please?

Kind Regards

Cyberkryption

jonasmalacofilho, 2020-01-05:
This has already been fixed in the repository (see PR #227, original issue #203).

There are a few other things to fixed before a new release is made, so until then you can use pyusb git.

```
pip install git+https://github.com/pyusb/pyusb
```

---

> **Installed via pip**
> **Current libraries installed**
> libusb (1.0.22b8) - Python binding for the libusb C library (ctypes/cffi-based libusb). INSTALLED: 1.0.22b8 (latest)
> pylibusb (0.2) - libusb wrapper (ctypes based)
> libusb1 (1.7.1) - Pure-python wrapper for libusb-1.0 INSTALLED: 1.7.1 (latest)

Note that neither of these are PyUSB.

Your logs do show that you're indeed using PyUSB, presumably version 1.0.2.

closed 2020-01-05 by jonasmalacofilho

cyberkryption, 2020-01-05:
Hi Jonas,

Thankyou very much.

I should have checked the PR,s but it was late.

Thankyou

Cybrrkryption

jonasmalacofilho, 2020-01-05:
Don't worry about it! : )