pyusb/pyusb #290: Error Accessing Strings on USB Stick - Win10 1903

cyberkryption, 2020-01-07:
Dear Maintainers,

I have pyusb working accessing the Vendor ID and Product ID for a USB memory stick but when it tries to access the serial number it shows "Error accessing a string" as shown below.

[image]

Could you tell me why this may be the case?

I looked at the closed issues and found #225 .Is this the same problem?

I have tried with several different USB sticks and get the same behaviour.

I tried the Nirsoft tool from https://www.nirsoft.net/utils/usb_devices_view.html and it can see the serial numbers.

I am using latest github version of pyusb under python 3.65 on Win 10 1903 with libusb 0.22 from the libusb.info website.

Any help appreciated.

Cyberkryption

jonasmalacofilho, 2020-01-07:
Yes, it seems to be the same issue, a driver vs. libusb limitation.

I'm not familiar with what the Nirsoft tool does to solve this.  If you find some info about that, please let me know.  Maybe it's using Usbdk...

closed 2020-01-18 by jonasmalacofilho

AndyChennnn, 2022-09-12:
I have the same problem, how can I solve this problem