pyusb/pyusb #291: usb.core.show_devices() return multiple instances of the same device

Wei1234c, 2020-01-16:
When I do `print(usb.core.show_devices(verbose = False))`  

which shows:
```

DEVICE ID 8087:0a2a on Bus 001 Address 019, Wireless Controller
DEVICE ID 0bda:2838 on Bus 001 Address 014, Specified at interface  <== same device 1 
DEVICE ID 0bda:2838 on Bus 001 Address 014, Specified at interface  <== same device 1 
DEVICE ID 0bda:2838 on Bus 001 Address 013, Specified at interface
DEVICE ID 046d:c52b on Bus 001 Address 002, Specified at interface  <== same device 2 
DEVICE ID 04f2:b57e on Bus 001 Address 003, Miscellaneous  <== same device 3
DEVICE ID 046d:c52b on Bus 001 Address 002, Specified at interface  <== same device 2 
DEVICE ID 04f2:b57e on Bus 001 Address 003, Miscellaneous  <== same device 3
```

There are multiple instances of the same device. Did I miss some configuration or it is a known issue?

jonasmalacofilho, 2020-01-16:
Hi!

Can you provide the output with `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug` set on the environment?

On what OS/platform is this, and what versions of Python, PyUSB and libusb (or other backend) are you using?

Wei1234c, 2020-01-17:
Sorry, I can't re-produce it. Somehow it's normal again. I don't know why.

Please close this issue. Sorry again for the interruption.

```
DEVICE ID 8087:0a2a on Bus 001 Address 002, Wireless Controller
DEVICE ID 0bda:2838 on Bus 001 Address 009, Specified at interface
DEVICE ID 0bda:2838 on Bus 001 Address 010, Specified at interface
DEVICE ID 046d:c52b on Bus 001 Address 001, Specified at interface
DEVICE ID 04f2:b57e on Bus 001 Address 003, Miscellaneous
```

jonasmalacofilho, 2020-01-18:
Ok : )

Please let me know if the problem reappears.

closed 2020-01-18 by jonasmalacofilho