pyusb/pyusb #292: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

leileigong, 2020-01-17:
I have two task.
the first on detect the plugin an plugout of pyusb device.the second one power on and power off the pyusb device.
when the device is pluged in. the first task will create another thread to operate the device.

the second task operation is
1.power on
2.power off
3.reboot device.
My code as follows:
`class CCIDDevice(object):
def init(self, dev):
self.dev = dev

def power_on(self):
    # set the active configuration. With no arguments, the first
    # configuration will be the active one
    self.dev.set_configuration()

    # get an endpoint instance
    cfg = self.dev.get_active_configuration()
    intf = cfg[(0, 0)]

    self.ep_out = usb.util.find_descriptor(
        intf,
        # match the first OUT endpoint
        custom_match=lambda e: \
            (usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT) and \
            (usb.util.endpoint_type(e.bmAttributes) == usb.util.ENDPOINT_TYPE_BULK)
        )
    assert self.ep_out is not None

    self.ep_in = usb.util.find_descriptor(
        intf,
        # match the first OUT endpoint
        custom_match=lambda e: \
            (usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_IN) and \
            (usb.util.endpoint_type(e.bmAttributes) == usb.util.ENDPOINT_TYPE_BULK)
        )
    assert self.ep_in is not None

    self.seq = 1

def power_off(self):
    usb.util.dispose_resources(self.dev)

def transmit_apdu(self, apdu):
    message = b'\x6F' + len(apdu).to_bytes(4, byteorder='little') + b'\x00' + \
              self.seq.to_bytes(1, byteorder='little') + b'\x00\x00\x00' + apdu
    # logging.debug(binascii.b2a_hex(message))
    self.ep_out.write(message)
    message = b''
    while True:
        packet = self.ep_in.read(self.ep_in.wMaxPacketSize, 12000)
        message += packet
        if len(packet) == self.ep_in.wMaxPacketSize:
            continue
        break
    # logging.debug(binascii.b2a_hex(message))
    self.seq += 1
    self.seq &= 0xFF

    return bytes(message[10:])`

But after run many times , the device was plug in/out many times , an error will accour.
the last time plugin  will got an error:
Traceback (most recent call last):
File "E:\FNTools\apps\FN70P\FN70PFWDL\DeviceDownWidget.py", line 632, in upgrade_firm
d.power_on()
File "E:\FNTools\fntkdevice\fntkdevice.py", line 70, in power_on
self.dev.power_on()
File "E:\FNTools\fntkdevice\usbccid.py", line 15, in power_on
self.dev.set_configuration()
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 869, in set_configuration
self._ctx.managed_set_configuration(self, configuration)
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 102, in wrapper
return f(self, *args, **kwargs)
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 147, in managed_set_configuration
self.managed_open()
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 102, in wrapper
return f(self, *args, **kwargs)
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\core.py", line 120, in managed_open
self.handle = self.backend.open_device(self.dev)
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
return _DeviceHandle(dev)
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 643, in init
_check(_lib.libusb_open(self.devid, byref(self.handle)))
File "C:\Users\lei.virtualenvs\FNTools-lKp0_7xw\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

jonasmalacofilho, 2020-01-18:
To me this seems to be inherently racy.  Between passing the device handle to the second thread and the call to `dev.set_configuration()`, the device can be disconnected again, no?

leileigong, 2020-01-19:
Do you mean call `usb.util.dispose_resources()` before `dev.set_configuration()`?

like this?
`    def power_on(self):
        usb.util.dispose_resources(self.dev)
        # set the active configuration. With no arguments, the first
        # configuration will be the active one
        self.dev.set_configuration()`

jonasmalacofilho, 2020-01-19:
No, I meant that to me it seems like this could be expected, if the device happens to be "plugged out" before the second thread reaches the failing `set_configuration` call.

The scenario you described appears to be inherently racy... what garantees do you have that the second thread can't use an invalid handle, besides the assumption that it will run quickly enough for device disconnections to not matter?

jonasmalacofilho, 2020-01-30:
Can you elaborate any further on how you're dealing with concurrency?

Specifically, how is concurrency managed between the thread that detects the connection of the device and opens a new handle to it, and the thread that receives that handle and communicates with the device using it?

closed 2020-02-16 by jonasmalacofilho

DrMeers, 2020-07-26:
I encountered this issue using antfs-cli -- and it turns out it was because I had the Garmin Express app open which was hogging the USB ANT+ connection -- once I closed that application, the `[Error 19] No such device` issue was resolved

mcuee, 2021-07-21:
Maybe the hotplug feature may help in this case. But take note libusb does not support hotplug under Windows yet.
https://github.com/pyusb/pyusb/pull/160