pyusb/pyusb #294: How to obtain manufacturers name

WouterJD, 2020-01-29:
Current code:
    devices = usb.core.find(find_all=True)
        for device in devices:
            print (device.bDeviceClass, hex(device.iManufacturer)

gives 0x4 for a Tacx Dongle

    print(device) shows 0x4 Tacx&VR (amongst the other output

How do I obtain the text "Tacx?VR" in my python program?

jonasmalacofilho, 2020-01-29:
You can use the device's [`.manufacturer`](https://github.com/pyusb/pyusb/blob/3bab67724ac4dc1d19f421abd302233314789b85/usb/core.py#L855-L861) property:

```py
print(device.manufacturer)
```

There are similar properties for the other standard string descriptors: `product` (for `iProduct`) and `serial_number` (for `iSerialNumber`).  Note: per the USB specification all string descriptors (even the standard ones) are optional.

Alternatively you can also use [`usb.util.get_string`](https://github.com/pyusb/pyusb/blob/3bab67724ac4dc1d19f421abd302233314789b85/usb/util.py#L272-L289) directly, if you're dealing with non-standard string descriptors or have a more complex use case.

WouterJD, 2020-01-29:
Thanks
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

closed 2020-01-29 by jonasmalacofilho