pyusb/pyusb #295: Device.serial_number returns None if 0

astrelsky, 2020-02-10:
If the serial number is 0 then the call to util.get_string returns None.

```
print(f'device: {self.dev.iSerialNumber}')
print(f'serial number: {self.dev.serial_number}')
```

results in
```
iSerialNumber: 0
serial_number: None
```

jonasmalacofilho, 2020-02-10:
This is expected: `iSerialNumber` is a **reference** to a string descriptor, and descriptor index 0 is used to indicate that (in this case) the device doesn't report a serial number (string).

> Zero is never the index of a real string. The USB spec allows a device to use zero in a string index field to indicate that no string is provided.  So the caller does not have to treat that case specially, this function returns None if passed an index of zero, and generates no traffic to the device.

closed 2020-02-10 by jonasmalacofilho

astrelsky, 2020-02-10:
> This is expected: `iSerialNumber` is a **reference** to a string descriptor, and descriptor index 0 is used to indicate that (in this case) the device doesn't report a serial number (string).

The documentation should reflect this then. I was expecting '0' and ended up having to use iSerialNumber directly.

jonasmalacofilho, 2020-02-11:
That quote was taken from the documentation.

More importantly, you probably shouldn't treat that device as having a serial number equal to `0` (string), because in reality it doesn't have any serial number.

Which is what `iSerialNumber == 0` means, according to the USB specification.  Though, admittedly, the spec isn't super clear on that...

astrelsky, 2020-02-11:
> That quote was taken from the documentation.
>
> More importantly, you probably shouldn't treat that device as having a serial number equal to `0` (string), because in reality it doesn't have any [...]

https://github.com/pyusb/pyusb/blob/bf88211d65caf01b903005efae7f0915a53a8c4e/usb/core.py#L829-L838

There is nothing about it being None there which is why I had expected '0'.
Regardless I've given up. I was looking to recreate https://github.com/raspberrypi/usbboot/blob/master/main.c but no matter what I did it always times out during writing even though I was doing the exact same thing. Ended up just running rpiboot with subprocess and moving on.