pyusb/pyusb #296: NotImplementedError: Operation not supported or unimplemented on this platform

premananda8, 2020-02-18:
I'm getting this error.

dev = usb.core.find(find_all=True,idVendor=0xX, idProduct=0xffff, backend=backend)
 for d in dev:
      print(d.serial_number)  #here getting NotImplementedError: Operation not supported or unimplemented on this platform

premananda8, 2020-02-18:
this library supports dfu devices?

jonasmalacofilho, 2020-02-18:
Please provide basic information about the problem:

 - what is the OS?
 - what is the product?
 - what backend are you using?
 - is there a kernel (aka device) driver installed for the device? 
 - what versions of Python and PyUSB are you using?

 ---

> this library supports dfu devices?

If it complies with the USB specification, then it's likely to be supported.  There are very very few exceptions, most notably devices that use isochronous transfer, which are not yet supported here.

jonasmalacofilho, 2020-02-19:
@premananda8 

I'm going to venture a guess that your issue has to do with Windows and a device that isn't a Human Interface Device (HID), nor has a libusb-compatible driver attached to it.

If that's indeed the case, take a look at [How to use libusb on Windows](https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows) on the libusb Wiki.

Please let me know.

premananda8, 2020-02-20:
Thanks, 
I installed zadig and my configurations as below:
OS:Windows7 64 bit
product:PID_25AD
backend:pyusb, package:usb.backend.libusb1
is there a kernel (aka device) driver installed for the device?:I installed zadig
what versions of Python and PyUSB are you using?:python 3.7 and pyusb:1.0.2

jonasmalacofilho, 2020-02-20:
> product:PID_FFFF

That's probably not a real PID... plus the VID is also necesasry.  Which device is it?

> is there a kernel (aka device) driver installed for the device?:I installed zadig

Which driver did you install to device _using_ Zadig?  And did it help?

premananda8, 2020-02-21:
VID:VID_25AD
PID:PID_FFFF
Actually it is in DFU mode , pyusb working fine when device is normal mode when making to DFU mode then using pyusb atleast i'm not getting serial_number as above code then not able to proceed also.

jonasmalacofilho, 2020-02-21:
In DFU mode the OS sees a completely different USB device.

So it is possible for libusb to be compatible with the driver for the device in normal mode (for example, if it's a HID), but not with driver for the device in DFU mode.

In fact, this is a common problem, and dfu-util does document that a libsub-compatible driver is needed on Windows (for the DFU mode):

> We offer binaries for Microsoft Windows and some other platforms. dfu-util uses libusb 1.0 to access your device, so on Windows you have to register the device with the WinUSB driver (alternatively libusb-win32 or libusbK), please see the libusb wiki for more details.
—http://dfu-util.sourceforge.net/

Using Zadig, try to install the WinUSB driver _for the device in DFU mode,_ and see if that helps.

premananda8, 2020-05-05:
not worked any solution for me, please help me if a device in dfu mode need to detect serial number, so that we can proceed upgrade to it

jonasmalacofilho, 2020-05-05:
Which driver is the device using in DFU mode?

premananda8, 2020-05-05:
it is smt device by using zadig i got extracted files from specific dfu device but i don't know how to use that dll files etc. in python

jonasmalacofilho, 2020-05-09:
Zadig will tell you which driver it has installed for a particular device.  Just remember that the device must first be set to DFU mode.

I've seen a problem with libusbK.dll (from Zadig) + PyUSB before, but I can't remember what it was.  Regardless, libusb-1-0.dll (not libusbK.dll) is at the very least preferred.

Please post the complete output of:

```
set PYUSB_DEBUG=debug
set LIBUSB_DEBUG=4
python <your test script>.py
```

premananda8, 2020-05-17:
[https://github.com/premananda8/usb/blob/master/dfu_devinfo.txt](url)

premananda8, 2020-05-17:
i want 'USB\VID_25AD&PID_FFFF\FA4521S.20000C2' line no.124 serial no. as FA4521S.20000C2

jonasmalacofilho, 2020-05-17:
```
[ 1.120064] [00000748] libusb: debug [libusb_open] open 1.2
[ 1.121064] [00000748] libusb: debug [winusb_open] unsupported API call for 'open' (unrecognized device driver)
[ 1.127064] [00000748] libusb: debug [libusb_open] open 1.2 returns -12
Operation not supported or unimplemented on this platform
```

Note: **unrecognized device driver**.

The issue appears to be that `USB\VID_25AD&PID_FFFF\FA4521S.20000C2` hasn't been configured to use a compatible driver.

I suppose it this could be the result of an issue in Zadig (or even some obscure libusb or Windows bug) but, if I'm honest, I think it's more likely that you simply missed something.  For exemple, the need to install driver once the device is (already) in DFU mode.

Regardless, this isn't a PyUSB issue and there isn't anything more I can do help... especially because after all this time you still haven't clearly answered which driver is in use (in DFU mode).

closed 2020-05-17 by jonasmalacofilho

premananda8, 2020-05-26:
without using zadig can i detect serial number in dfu mode