pyusb/pyusb #298: Version number is out-of-sync with Git code

eblot, 2020-02-25:
Hi,

I'm the maintainer of PyFtdi. I had some hard time figuring out why some reported issues with PyFtdi depicting PyUSB traces did not match at all my own installation of PyUSB. Even after reinstalling from scratch, clearing out pip caches, etc, there were obvious discrepancies between packaged 1.0.2 version available from PyPi and the traces I was reading.

PyUSB has resumed development and that's really great.

However the version number has not been updated and is still defined as `1.0.2`. Current delta with `1.0.2` is:  `30 files changed, 775 insertions(+), 630 deletions(-)`. This is really huge.

It would be nice to add a -dev [or any common suffix](https://www.python.org/dev/peps/pep-0440/#pre-releases) to the `__init__.py` file so that there is no confusion with the active development version and the packaged version on PyPi. I guess PyFtdi will not be the only package which depends on PyUSB and will encounter some maintenance issues to deal with, and confusion is likely to occur.

Note: I used to push PyUSB for the original author (Wander) to [PyPi](https://pypi.org/project/pyusb/#history), so if you do not have access yet to PyPi for PyUSB, please contact me.

Thanks.

jonasmalacofilho, 2020-02-25:
Hi again,

I'll consider doing this after the 1.1.0 release.

But I haven't seen this work particularly well in my own stuff: it still is a static version number shared by many commits.

> Additional note: it seems you have not installed PyUSB from an official package but a development version.
—_(from the downstream issue)_

Regarding stable vs. development releases, the most common cases (pip install git or --editable) can be handled by asking users to provide the output of `pip list`.

Nonetheless, I'll give it some (thorough) thought.

eblot, 2020-02-25:
Well, I agree it is not perfect, at least it gives an instant clue from the traceback *e.g.* `site-packages\pyusb-1.0.2-dev-py3.6.egg` vs. `site-packages\pyusb-1.0.2-py3.6.egg`., i.e. is it a release or any development version.

To avoid static version, you may add the git distance from the last official tag:

    $ git log -1
    commit cf88dfcd3b38afd00ec9fc7229213c8e8aa42118 (HEAD -> master, origin/master, origin/HEAD)
    Author: root <572488191@qq.com>
    Date:   Sun Feb 23 20:06:06 2020 +0800
    
        Remove non-ASCII dashes from copyright notices (#297)
    
        In Python 2 source files are treated as encoded in ASCII.
    eblot@spey:~/Sources/Git/github.com/pyusb
    
    $ git describe
    v1.0.2-38-gcf88dfc
    eblot@spey:~/Sources/Git/github.com/pyusb
    
    $ git pull
    From github.com:pyusb/pyusb
       cf88dfc..998e1da  master     -> origin/master
    Updating cf88dfc..998e1da
    Fast-forward
     usb/backend/libusb0.py | 4 ++++
     1 file changed, 4 insertions(+)
    eblot@spey:~/Sources/Git/github.com/pyusb
    
    $ git describe
    v1.0.2-39-g998e1da

ap--, 2020-06-03:
If it would help, I can work on a PR for integrating [setuptools_scm](https://github.com/pypa/setuptools_scm) which would handle the versioning in an automated way.

@jonasmalacofilho @mcuee Thanks for the work you put into `pyusb`, it's great to see that it's continued to be developed :smiley: 

Cheers,
Andreas

jonasmalacofilho, 2020-06-05:
Hi @ap--,

First, I really appreciate you volunteering to do this work!  It took me a while to organize my thoughts on this, but here they are:

I really like your solution (even though it won't be foolproof, because no such thing is possible here),¹ but I see a problem with the extra dependency.

It seems that this may break in systems with old versions of setuptools, one way (toml file) or the other (setup_requires).  This would particularly affect Linux distributions packaging PyUSB.

Can you check that stuff like [this bug](https://lists.gt.net/python/python/1408890) is no longer an issue on any distro versions that would ship a 1.1.0 PyUSB?  Besides Debian/Ubuntu, RHEL and CentOS are also likely candidates for problems.

In summary: as long as this is packageable in the major Linux distros, I'm fine with it.  
_(I think the usual avenues for Windows (manual or Conda) and Mac (Homebrew) should be able to handle this without problems)._

On my side, this will also require a bump on the minor version number.

---

_¹ The base version number has some meaning, but the truth is that users may always be running completely random code.  Either intentionally or by accident.

_For an extreme example, some time ago a company working with mine on a project was having meritorious issues compiling our project.  Eventually we figured out that, unknown to them, they were running a completely borked standard library (for the particular language we were using), which even included modules from different  _major_ (and totally incompatible) versions of the language!  How do you catch this with a version string?_

ap--, 2020-06-07:
Hi @jonasmalacofilho,

Great :smiley: 
I'll start working on it as soon as i have some free time and will open a PR where we can discuss further.
I have experience with packaging for pypi and conda for all OSes, but so far my experience with packaging deb/rpms/etc. for different Linux distributions is limited to using [fpm](https://github.com/jordansissel/fpm) once. But I'm happy to dive deeper into the topic :+1: 

Cheers,
Andreas

closed 2020-06-28 by jonasmalacofilho