pyusb/pyusb #299: Use of `usb_get_driver_np` may trigger unexpected exception

eblot, 2020-02-25:
I'm not sure I figured out the whole picture, but call to this OS-specific version has been recently added to `is_kernel_driver_active` in `libusb0` backend.

This function may not be implemented, i.e. not available in some shared libraries, depending on the OS/build [options](https://github.com/libusb/libusb-compat-0.1/blob/master/libusb/usb.h).

When `is_kernel_driver_active` is called:
 * if `libusb1` backend is used, the portable `libusb_kernel_driver_active` function is invoked. On non-Linux platforms, `LIBUSB_ERROR_NOT_SUPPORTED` is returned, which is translated through `_check()` into a `NotImplementedError` exception.
 * if `libusb0` backend is used, the [non-portable](https://sourceforge.net/p/libusb-win32/wiki/Documentation/) `usb_get_driver_np` function is called. On non-Linux platforms, ctypes fails to find the symbol - as it does not exist in the underlying shared library, which directly translates into an opaque `AttributeError` exception, exposing pyusb internals but giving little clue about the actual error:
    `AttributeError: function 'usb_get_driver_np' not found`

Calling `is_kernel_driver_active` from a PyUSB client and prepare for catching the `NotImplementedError` exception if the OS does not support is an easy, portable way to get access to a USB device.

However, with the current implementation, it is no longer possible as the actual behaviour depends on the combination of the OS and the selected libusb backend.

I think it would be nice to enclose the `usb_get_driver_np` call into an exception catcher and raise a `NotImplementedError ` exception if this non-portable function is not available from the loaded libusb instance.

The same paradigm would also apply for all non-portable function such as `usb_detach_kernel_driver_np`, as public PyUSB API methods ends up directly calling non-existing functions on `libusb` with `libusb0` backend. This issue does not exist in `libusb1` backend, as no non-portable functions are used with this backend.

Note: up to official *v1.0.2*, `is_kernel_driver_active` was directly managed by the backend wrapper, so got never called with a `libusb0` backend. Now that is has been partially implemented in `libusb0` backend, I would suggest to protect its usage when the libusb API functions are not implemented/exposed.

jonasmalacofilho, 2020-02-25:
Thank you for the detailed bug report.

It should be fixed now, though I still need to test it on Windows.

closed 2020-02-25 by jonasmalacofilho

eblot, 2020-02-25:
Thanks for the quick fix.

cederom, 2021-05-13:
This bug still exists and is valid on FreeBSD as reported here: https://github.com/pyusb/pyusb/issues/365

jonasmalacofilho, 2021-05-13:
@cederom 

It's not the same bug.  In fact, on platforms where ctypes fails to find the `usb_get_driver_np` symbol, like the case described by eblot, `is_kernel_driver_active()` will raise `NotImplementedError` as intended.

The problem is that we didn't anticipate that `usb_get_driver_np` could be available on some platforms, yet with slightly (but significantly) different semantics.

commit 998e1da0 (2020-02-25): Prevent AttributeErrors from non-portable libusb0 functions

usb_get_driver_np and usb_detach_kernel_driver_np are marked as
non-portable and may be unavailable; for example, they are not exported
by libusb-win32.

To make both libusb0 and libusb1 consistent accross all platforms
prevent the AttributeError and instead raise a NotImplementedError.

P.S. Windows-only functions already were correctly handled.

Reported-by: @eblot
Fixes: #299 ("Use of `usb_get_driver_np` may trigger unexpected exception")