pyusb/pyusb #300: Poor performance while sending data streams

emob-nordic, 2020-03-13:
I wanted to use pyusb to test my device.  For this i need to have a possibility to write to an endpoint every SOF (1ms). What i encounter is very slow write performance while using pyusb. Timeout needed for successful write must be ~>16ms which is sad.

Here comes my question:
Is there is any chance for improvements for sending data continuously?

I am enclosing screen shoot from USB logic analyzer.
[image]

closed 2020-03-13 by emob-nordic

jonasmalacofilho, 2020-03-13:
Why did you already close this?

rabryan, 2020-03-13:
He probably did it by accident because the github app can be confusing for EEs🙂

emob-nordic, 2020-03-16:
I closed this on purpose. Problem was (as usual) between the chair and the screen ;) I was trying to send each chunk of data separately, each time calling:
```
dev.write(ep=0x??, data=<list_of_raw_data_chunk>, timeout)
```
I tested with more data at once and it works well.
There are still some concerns I have. In particular:
- I've checked that maximum payload of one write() is 128*<ep_size>, why? is it likely to change?
- I am unable to send and receive at the same time, is it any way to do it? I've tried using threading but it didn't work out.

I am wondering i I should open a new issue regards points above?

jonasmalacofilho, 2020-03-16:
Hi @emob-nordic!

That makes sense, as it added significantly more overhead.

Still, how much data at once do you need to send to reach your < 16 ms target?  (looking to see if there isn't too much overhead somewhere)

> I've checked that maximum payload of one write() is 128*<ep_size>, why? is it likely to change?

I'll have to check that.  Can tell me how/where you found this limitation, or supply a code sample that reproduces it?

> I am unable to send and receive at the same time, is it any way to do it? I've tried using threading but it didn't work out.

What exactly do you mean by _at the same time?_  I ask because (I think that) at the lower levels the transfers will always be serialized.

On higher levels it should be possible to start multiple transfers _roughly_ at the same time (e.g enqueue the next read while the current write is still being sent), but there will probably be a mutex in libusb and/or kernel.

---

Please feel free to open issues for these points.

emob-nordic, 2020-03-16:
First of all brief clarification what I am doing.
I have an audio device that copies data form OUT to IN endpoint. 
Endpoints are: 0x08 (OUT) , 0x88 (IN). What I want to do is verification if:
data send from host to OUT ep are 1:1 received on IN ep.

My device is configured as follows:
- sampling frequency: 48kHz
- channels: 2
- bitResolution: 16b (2B)

Taking into consideration that data are send every 1ms frame interval. Endpoint has to have at least:
48000(sampling_freq)*2(bitRes)*2(channels)/1000(frames/s) = 192 size.

>Still, how much data at once do you need to send to reach your < 16 ms target? (looking to see if there isn't too much overhead somewhere)

I want to stream data for 1sec which is 192000 Bytes (assuming previous callculations above).

I've checked that calling write() has ~16-17ms overhead . I've checked that experimentally:
```
write(ep, 192Bytes of data, timeout > 17)
write(ep, 192Bytes of data, timeout > 17)
```
and I saw 16-17 SOFs in between (see attached pic in the issue above) so I assume this is the overhead.

> I'll have to check that. Can tell me how/where you found this limitation, or supply a code sample that reproduces it?

Assuming in my case ep_size = 192. I've tried to send as much as I can at once to be able to send chunks of data each SOF (1ms). Then I realized it is possible to send max 128*128. If I try to put more data I get: 
```[Errno 22] Invalid parameter ``` dunno where the limitation is as I did not check the internals.

> What exactly do you mean by at the same time? I ask because (I think that) at the lower levels the transfers will always be serialized.

What I mean by that is I want to send 1sec of recording (192000 raw Bytes) and at the same time be able to read them in other thread). Right now, even I am using Thread, the code first sends the data and then tries to receive. See the pic below from USB analyzer. Please look closely at timestamps. The IN token was issued ~17ms after OUT transaction has ended it means write() read() calls are sync. I need async calls to work with that.

[image]

emob-nordic, 2020-03-16:
An example code I am using: (keep in mind I am not Python expert ;p ) 
```
amount = int(192*1)
timeouts = 17 + int(amount/192) ## 16+1 for safety is overhead for sending,
#assuming same overhead for receive

def prepare(dev):
        for cfg in dev:
                for intf in cfg:
                        if dev.is_kernel_driver_active(intf.bInterfaceNumber):
                                print(intf.bInterfaceNumber)
                                try:
                                        dev.detach_kernel_driver(intf.bInterfaceNumber)
                                        log_dbg("Kepernel driver detached")
                                except usb.core.USBError as e:
                                        sys.exit("Could not detach kernel driver form iface {}".format(e))

        cfg = dev.get_active_configuration()
        if cfg is None:
                log_dbg("No cfg, exit")
                exit(1)

        try:
                dev.set_configuration(1)
        except usb.core.USBError as e:
                sys.exit("Set configuration failed {}".format(e))

        try:
                # set alternate 2 to active (able to receive)
                dev.set_interface_altsetting(1,1)
        except usb.core.USBError as e:
                sys.exit("Error {}".format(e))

        try:
                # set alternate 2 to active (able to send)
                dev.set_interface_altsetting(2,1)
        except usb.core.USBError as e:
                sys.exit("Error {}".format(e))

def read_thread(dev):
        log_dbg("reading thread started")
        try:
                dev.read(0x88, len(list(raw_data[0:amount])), timeout=timeouts)
                time.sleep(0)
        except usb.core.USBError as e:
                sys.exit("Error on read {}".format(e))

def send_thread(dev):
        log_dbg("sending thread started")
        try:
                dev.write(endpoint=8, data=list(raw_data[0:amount]), timeout=timeouts)
                time.sleep(0)
        except usb.core.USBError as e:
                sys.exit("Error on send {}".format(e))

dev = usb.core.find(idVendor=VID, idProduct=PID)
if dev is None:
        sys.exit("Device not found, exit")

prepare(dev)

Thread(target=send_thread(dev)).start()
Thread(target=read_thread(dev)).start()

```

rabryan, 2020-03-16:
Yes it is

mcuee, 2021-08-02:
Further discussion here: https://github.com/pyusb/pyusb/issues/301

mcuee, 2021-08-02:
From #301, actually the issue may have to do with isoc support with pyusb.