pyusb/pyusb #301: Send/Read at the same time is not possible.

emob-nordic, 2020-03-16:
This issue is follow up of #300 and is similar to #226
I am using pyUSB to verify if data stream is correct on my device. My device is an audio headset with 2 endpoints, IN and OUT. Data from OUT ep are copied to IN ep. I want to verify if this happens correctly. Simply speaking if data that Host send to an OUT ep are later received on OUT ep.

I've prepared sample code which runs 2 threads:
- one for sending
- one for reading

**There are 2 issues I've found:**
1) pyUSB does not support reading & receiving at the same time
The problem is those threads does not work cooperative with pyUSB backend (probably). I was expecting this logic sequence:
```
start send thread
start read thread
...
OUT transaction (send)
IN transaction (read)
OUT transaction (send)
IN transaction (read)
OUT transaction (send)
IN transaction (read)
...
```

What i see:
```
OUT transaction (send)
OUT transaction (send)
OUT transaction (send)
...
IN transaction (read)
IN transaction (read)
IN transaction (read)
...
```
It seems that read/write is blocking the backend. Is there any chance to implement asynchronous read/write?

2) pyUSB has limitation to number of packets that could be send with one write().
I want to send data that are sine wave with:

    - sampling frequency: 48kHz
    - channels: 2
    - bitResolution: 16b (2B)

Taking into consideration that data are send every 1ms frame interval. Data are send in packets with 48000(sampling_freq)*2(bitRes)*2(channels)/1000(frames/s) = 192B size.

Maximal data length I am able to send with one write() is: 192* **128**. It means 128 OUT transaction will happen. If I try to send 192* **129** -> ```[Errno 22] Invalid parameter```

I am enclosing sample code + screenshoot from logic analyzer.

[image]

The file with the code (note Im not a python expert - this is just script for testing)
[test.txt](https://github.com/pyusb/pyusb/files/4339228/test.txt)

emob-nordic, 2020-03-18:
I've checked that it is for sure not the problem with libusb which I am using.
```write()``` function in the end submits a transfer to the pyUSB [backed](https://github.com/pyusb/pyusb/blob/cf88dfcd3b38afd00ec9fc7229213c8e8aa42118/usb/backend/libusb1.py#L680). This call is not blocking as You can [read](http://libusb.sourceforge.net/api-1.0/group__libusb__asyncio.html#gabb0932601f2c7dad2fee4b27962848ce).

It seems that the thing that blocks is the [while loop](https://github.com/pyusb/pyusb/blob/cf88dfcd3b38afd00ec9fc7229213c8e8aa42118/usb/backend/libusb1.py#L682) just after the submission. It is pooling for the events from the USB backed. Are there any plans to rework and make pyUSB non blocking?

mcuee, 2020-06-02:
I believe this will be addressed in ticket #63.

mcuee, 2021-08-02:
@emob-nordic Actually you are using isoc transfer here, which is kind of async in nature. I myself do not have good test results with isoc transfer with pyusb, at least under Windows and macOS. Which OS are you using? 

It is not really easy to get isoc transfer correctly done, and libusb itself may have problems with isoc transfer under Windows.