pyusb/pyusb #302: How to reach useful member of the resource manager?

eblot, 2020-03-19:
The `_ResourceManager` instance is a private member of the `Device`, *i.e.* `dev._ctx`

It is therefore not supposed to be called from a PyUsb client.
However I did not find a public API way to test whether a connection is opened, so I rely on checking `dev._ctx.handle`. Is there any clean way to check this?

I noticed that even `usb.util` access this private `Device` member.

Another related question: how to call `backend.reset_device()` from a `Device` instance (to avoid going through the `_ResourceManager`, *e.g.* `dev._ctx.backend.reset_device(dev._ctx.handle)`

Thanks.

jonasmalacofilho, 2020-05-09:
Hi,

> However I did not find a public API way to test whether a connection is opened, so I rely on checking dev._ctx.handle. Is there any clean way to check this?

Not at the moment, but you might be able to skip the check althoghether.

For example, instead of checking whether there's a handle open before calling `usb.util.dispose_resources`, simply call it directly: there's no (functional) harm in calling it for a device with no open handle.

> Another related question: how to call backend.reset_device() from a Device instance (to avoid going through the _ResourceManager, e.g. dev._ctx.backend.reset_device(dev._ctx.handle)

There's `Device.reset()`:

https://github.com/pyusb/pyusb/blob/195c89f974b79cb045bbbee1ef04cd00282ccffb/usb/core.py#L945-L950

Let me know if this helps.

eblot, 2020-05-09:
> you might be able to skip the check althoghether.

Unfortunately not. There are some conditions where the resource manager gets in the way, if I may say so. If the connection has been closed, the resource manager will force a re-open and this may trigger a very bad situation: libusb natively crashes, so the Python interpreter segfaults.

Therefore, no single call to libusb should be performed if the connection has been closed, or the whole interpreter is killed. Testing the handle is a functional workaround.

> `Device.reset()` 

It does go through the context manager (`self._ctx`)...

jonasmalacofilho, 2020-05-09:
> There are some conditions where the resource manager gets in the way, if I may say so. If the connection has been closed, the resource manager will force a re-open and this may trigger a very bad situation: libusb natively crashes, so the Python interpreter segfaults.

Can you expand on what exactly are those conditions, as they appear to be serious bugs?

> Therefore, no single call to libusb should be performed if the connection has been closed, or the whole interpreter is killed. Testing the handle is a functional workaround.

PyUSB should be able reopen any handle (as long as the device still exists on the OS level, but having an open handle doesn't guarantee that anyway) that's necessary (and only if it's necessary), thus not causing the interpreter to be killed.

> It does go through the context manager (self._ctx)...

Yes, but I don't see any problem in that.  You need an open handle to reset a device, and calling `dispose` also seems pertinent or the manager could be left in an incoherent state.

eblot, 2020-05-09:
It will be hard for me to remember as I bumped into this issue months ago, but some conditions, pyusb attempts (through `managed_open()`) to re-open a connection which has been closed and crash libusb. I really do not remember how to trigger this issue, I'll try to reproduce it.

I think it is related to exiting the Python interpreter. For some reason, libusb may have started to release the USB connection when managed_open() kicks in and try to reopen the connection to the same device while libusb as discarded it. I do not remember the exact sequence, but I remember it happens *sometimes* when the interpreter leaves. Some  sort of race conditions between pyusb and libusb. Note: I do no use some kind of `atexit` call, I triple checked this :-)

eblot, 2020-05-09:
> Yes, but I don't see any problem in that.
It is to avoid the `managed_open()` call for the same reason as above.

jonasmalacofilho, 2020-05-09:
> I think it is related to exiting the Python interpreter.

Any chance this happened on Windows, with libusb >= 1.0.22?

> It is to avoid the managed_open() call for the same reason as above.

Ok, that makes sense, but then you wouldn't call `reset_device` anyway.  So the issue really is `managed_open()` potentially crashing.

eblot, 2020-05-09:
Mostly on macOS, maybe on Linux (I cannot remember). I try to stay away from Windows as strong as I can :-)

jonasmalacofilho, 2020-05-09:
Well, the bug that caused visible problems in Windows was more general and, in theory, could also affect other platforms, maybe just not as frequently.

I'll take a look and see if I can spot anything still obviously wrong.

If you try to reproduce the problem, see if it's somehow dependent on the fix in #227 ("Fix #203: libusb sometimes cleaned up too early").

> I try to stay away from Windows as strong as I can :-)

Me too, if I'm honest.

eblot, 2020-05-09:
#227:  ok I will. For now, I'm unable to reproduce the issue but I remember it was not 100% reproducible. 
Note that I use tagged version 1.0.2 which is quite old.

eblot, 2020-05-09:
#203/#227 is indeed quite interesting.

mcuee, 2020-12-24:
@eblot: can we close this issue for now with the new version released? You can always reopen if you encounter issues again.

eblot, 2020-12-24:
Yes you may close it. I'll keep calling the private interface anyway as many people do not use a recent version of pyusb.

closed 2021-01-19 by jonasmalacofilho