pyusb/pyusb #304: to check usb camera is connected or not?

Ritesh1991, 2020-05-04:
can you tell me if we connect two or more camera with linux ubuntu16.04 and than i wan to check camera connection than what should i need to do? And how to use this library to check camera connection?

premananda8, 2020-05-05:
try this example
[https://github.com/premananda8/python/blob/master/usb_detect.py](url)

jonasmalacofilho, 2020-05-05:
> i wan to check camera connection than what should i need to do?

It depends on what you mean by _checking_ and _camera connection._

At the simplest, `lsusb` should suffice.  If I remember correctly in Ubuntu it's part of the `usbutils` package.

---

> And how to use this library to check camera connection?

Following an equivalent interpretation of checking/connection as before:

```py
import usb

for dev in usb.core.find(find_all=True):
    addr_part = f'Bus {dev.bus:03} Device {dev.address:03}:'
    id_part = f'ID {dev.idVendor:04x}:{dev.idProduct:04x}'
    names_part = []
    try:
        if dev.manufacturer:
            names_part.append(dev.manufacturer)
        names_part.append(dev.product)
    except ValueError:
        # can't access descriptors (permission error?) or dev doesn't have them
        pass
    print(addr_part, id_part, *names_part)
```

For more information, take a look at the [tutorial](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst), as well as the documentation for `core`, specially `find` and `Device`:

```
$ pydoc usb.core
$ pydoc usb.core.find
$ pydoc usb.core.Device
```

closed 2020-05-09 by jonasmalacofilho