pyusb/pyusb #305: Invalid endpoint

Prolasis, 2020-05-07:
I am trying to send UVC commands to a UVC camera to move the pan tilt. So far my code works up to write / sending the commands. I get an error saying invalid endpoint. even though i can debug and see the ep set the bEndpointAddress to 129. Any thoughts? what i am doing wrong here

`import usb.core
import sys
import usb.util

dev = usb.core.find(idVendor=0x2E7E, idProduct=0x0601)
if dev is None:
    raise ValueError('Our device is not connected')
else:
    print("found")
    print("device class is " + str(dev.bDeviceClass) + "   number of device config is "+ str(dev.bNumConfigurations))
for cfg in dev:
    sys.stdout.write(str(cfg.bConfigurationValue) + '\n')

    for intf in cfg:
        sys.stdout.write('\t' + \
                         str(intf.bInterfaceNumber) + \
                         ',' + \
                         str(intf.bAlternateSetting) + \
                         '\n')
        for ep in intf:
            sys.stdout.write('\t\t' + \
                             str(ep.bEndpointAddress) + \
                             '\n')

cfg = dev[0]
intf = cfg[(0,0)]
ep = intf[0]

dev.set_configuration() # we assume the configuration 5 is the first one

USBSendCommand = '81 01 06 01 VV WW 02 03 FF'

cfg.set()
dev.set_configuration(cfg)

dev.write(129,USBSendCommand)`

Prolasis, 2020-05-07:
`import usb.core
import sys
import usb.util

dev = usb.core.find(idVendor=0x2E7E, idProduct=0x0601)
if dev is None:
    raise ValueError('Our device is not connected')
else:
    print("found")
    print("device class is " + str(dev.bDeviceClass) + "   number of device config is "+ str(dev.bNumConfigurations))
for cfg in dev:
    sys.stdout.write(str(cfg.bConfigurationValue) + '\n')

    for intf in cfg:
        sys.stdout.write('\t' + \
                         str(intf.bInterfaceNumber) + \
                         ',' + \
                         str(intf.bAlternateSetting) + \
                         '\n')
        for ep in intf:
            sys.stdout.write('\t\t' + \
                             str(ep.bEndpointAddress) + \
                             '\n')

cfg = dev[0]
intf = cfg[(0,0)]
ep = intf[0]

dev.set_configuration() # we assume the configuration 5 is the first one

USBSendCommand = '81 01 06 01 VV WW 02 03 FF'

cfg.set()
dev.set_configuration(cfg)

dev.write(129,USBSendCommand)`

jonasmalacofilho, 2020-05-09:
Endpoint `129`/`0x81`/`0b10000001` is never a valid endpoint **to write to**:

[image]
_USB 3.2 Specification (Revision 1.0): Table 9-25: Standard Endpoint Descriptor_

closed 2020-05-09 by jonasmalacofilho