pyusb/pyusb #310: How to claim more than one USB interface ?

ehabhussein, 2020-06-27:
Is there a way to claim more than one USB interface at a time ?

jonasmalacofilho, 2020-06-27:
I could be wrong, but this should already work as is.  Both with automatically/as needed or using the explicit methods in `usb.util`.

ehabhussein, 2020-06-27:
Do the maintainers mind to share some code here on how to accomplish this ?

jonasmalacofilho, 2020-06-27:
Sorry, I simply can't test this right now.

Also, [you already know how to claim one interface](https://github.com/ehabhussein/AutoGadgetFS/blob/master/libagfs.py#L282).  Try to do the same with the second interface¹ and, in case that fails, report it here² and we can try to investigate why that happens.

```py
usb.util.claim_interface(device, interface1)
usb.util.claim_interface(device, interface2)
```

Notes:

_¹ Assuming its from the same configuration, or this simply isn't possible according to the USB spec._  
_² Closing this for now; if it later turns out there's a bug to fixed, I'll reopen it._

closed 2020-06-27 by jonasmalacofilho

ehabhussein, 2020-06-27:
yes i tried and it works only on the same config obviously. Thanks

jonasmalacofilho, 2020-06-27:
Glad to hear, and thanks for letting me know it does indeed work.

By the way, AGFS seems interesting.  I'll give it a better look when I have some time.

ehabhussein, 2020-06-27:
thank you! ill be presenting it at blackhat this year and it wouldn't have been possible if it weren't for PyUSB thank you all for the amazing work!

jonasmalacofilho, 2020-06-27:
Awesome!  Do you mind at least sending me the slides after your talk?  I'll try to find video once it's publicly released, but I'm not sure how long this usually takes with Blackhat, nor if all talks are made available.

---

I appreciate the support, although merit should go to @walac.  I'm just the new maintainer...

ehabhussein, 2020-06-27:
Will do definitely