pyusb/pyusb #311: Could puysb support UFI command?

hyhezhen, 2020-07-11:
I want to send  "INQUIRY" command to get some information from usb disk. And I don't know how to...

jonasmalacofilho, 2020-07-11:
PyUSB should allow you to do that, yes.

You'll need to read the [UFI Command Specification](https://usb.org/sites/default/files/usbmass-ufi10.pdf) (plus figure out whether you need to use [Bulk-Only Transport](https://www.usb.org/sites/default/files/usbmassbulk_10.pdf) or [Control/Bulk/Interrupt Transport](https://usb.org/sites/default/files/usb_msc_cbi_1.1.pdf)) and write the appropriate low-level request(s) to the device.

I don't have experience working with mass storage class devices, but I'll do my best to help you if you have any specific questions along the way.

mcuee, 2020-07-15:
Example from libusb may be of some reference value.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L316

hyhezhen, 2020-07-19:
Very Thank for jonasmalacofilho  and  mcuee.

And It  seemed  that  the code   in
https://github.com/lewurm/ppcskel/blob/master/usb/drivers/class/storage.c  
is easier for pyusb.

I only want to  inquiry,like this:
>>>
u8 usb_storage_inquiry(u8 device)
{
	/* send cwb "usbc" */
	
	usb_storage_cbw *cbw = (usb_storage_cbw*)malloc(sizeof(usb_storage_cbw));
	cbw->dCBWSignature= 0x43425355;
	cbw->dCBWTag=0x826A6008;
	cbw->dCBWDataTransferLength=0x00000024;
	cbw->bCWDFlags=0x80;
	cbw->bCBWLun=0x00;
	cbw->bCBWCBLength=0x01;

	u8 i;
	for(i=0;i<16;i++)
		cbw->CBWCB[i]=0x00;

	cbw->CBWCB[0]=0x12; // 0x12 = INQUIRY

	usb_bulk_write(massstorage[device], 2, (u8*)cbw, 31, 0); 
	usb_bulk_read(massstorage[device], 1, (u8*)cbw, 36, 0); 
	usb_bulk_read(massstorage[device], 1, (u8*)cbw, 13, 0); 

	free(cbw);

	return 0;
}
===========================================================

===================================================================================   
So I add code in core.py

def bulk_write(self, ep, intf,
            data_or_wLength = None, timeout = None):
        
        try:
            buff = util.create_buffer(data_or_wLength)
        except TypeError:
            buff = _interop.as_array(data_or_wLength)

        self._ctx.managed_open()

        print(self.__get_timeout(timeout))
        ret = self._ctx.backend.bulk_write(
                                    self._ctx.handle,
                                    ep,
                                    intf,
                                    buff,
                                    self.__get_timeout(timeout))

 def bulk_read(self, ep, intf,
            data_or_wLength = None, timeout = None):

        try:
            buff = util.create_buffer(data_or_wLength)
        except TypeError:
            buff = _interop.as_array(data_or_wLength)

        self._ctx.managed_open()

        print(self.__get_timeout(timeout))
        ret = self._ctx.backend.bulk_read(
                                    self._ctx.handle,
                                    ep,
                                    intf,
                                    buff,
                                    self.__get_timeout(timeout))

=============================================================================== 
===================================================================================In my code:
import usb.core
import usb.util
from ctypes import *

class command_block_wrapper:
     _fields_ = [('dCBWSignature', c_uint32),
                ('dCBWTag', c_uint32),
                ('dCBWDataTransferLength', c_uint32),
                ('bmCBWFlags', c_uint8),
                ('bCBWLUN', c_uint8),
                ('bCBWCBLength', c_uint8),
                ('CBWCB', c_uint8 * 16),
                ]

cbw = command_block_wrapper()
cbw.dCBWSignature = b'\x43\x42\x53\x55'
cbw.dCBWTag=0x826A6008
cbw.dCBWDataTransferLength=0x00000024
cbw.bCWDFlags=0x80
cbw.bCBWLun=0x00
cbw.bCBWCBLength=0x01

cbw.CBWCB = 0x12

idV = 0x04b4
idP = 0xde01
# find our device
dev = usb.core.find(find_all=False,idVendor=idV, idProduct=idP)

if dev is None:
    raise ValueError('Device not found')
    
dev.set_configuration()

cfg = dev.get_active_configuration()
intf = cfg[(0,0)]

ep_out = usb.util.find_descriptor(
    intf,
    
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)
        

ep_in = usb.util.find_descriptor(
    intf,
    # match the first IN endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_IN)

dev.bulk_write(ep_out,intf,(cbw))         ???
dev.bulk_read(ep_in,intf,cbw)              ???
dev.bulk_read(ep_in,intf,cbw)              ???

I don't to known how to write the   last three lines of code (end with ???)

Could  you help me on it.     Thanks

jonasmalacofilho, 2020-07-20:
You need to convert your `command_block_wrapper` into an `array('B', ...`..  After that, `Device.write` can automatically figure out when to use bulk writes (and reads) from the endpoint.

```python
dev.write(ep_out, cwb.toarray())  # passing `bytes` or `list` are ok too, but will be converted internally
recv1 = dev.read(ep_in, 36)
recv2 = dev.read(ep_in, 13)
```

If you want, you can try to reuse `recv1` to avoid another buffer allocation in the second `read`.  But I'm guessing you actually need the data for something else.

hyhezhen, 2020-07-23:
Thanks to jonasmalacofilho。
I am more familiar with C  language,  and the  C  code is ok.

ret = usb_bulk_write(dev, EP_OUT, (u8*)cbw, 31, 1000); 
ret = usb_bulk_read(dev, EP_IN, (u8*)buffer, 64, 1000); 

Yes,you're right,I don't need to write my "bulk_write".

dev.write()       dev.read()    is  ok.

I change calittle for the class:

I change calittle for the class:
class command_block_wrapper(LittleEndianStructure):
    _pack_ = 1
    _fields_ = [('dCBWSignature', c_uint32),
                ('dCBWTag', c_uint32),
                ('dCBWDataTransferLength', c_uint32),
                ('bCWDFlags', c_uint8),
                ('bCBWLUN', c_uint8),
                ('bCBWCBLength', c_uint8),
                ('CBWCB', c_uint8 * 16),
    ]

    def encode(self):
        return string_at(addressof(self), sizeof(self))

    def decode(self, data):
        memmove(addressof(self), data, sizeof(self))
        return len(data)
        
        
cbw = command_block_wrapper()
....// set byte value to cbw

mycmd =  cbw.encode()
num = dev.write(ep,mycmd, 32)

ret  = dev.read(ep_in, 36)
print(ret)

=================================
Sometimes it print the result I want.  
But  more times  it show me:
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: 
I / O operation aborted ...
        
Restart the pc, re plugin in the usb device is useless.

hyhezhen, 2020-08-04:
Add  code  

if dev.is_kernel_driver_active(0):
    reattach = True
    dev.detach_kernel_driver(0)

before 
dev.set_configuration()

should be better.

closed 2020-08-04 by hyhezhen