pyusb/pyusb #313: usb.core.USBError: [Errno 5] Input/Output Error

WanManito, 2020-08-12:
[image]

The previous query device did not have any exceptions，
When I execute write, the exception is reported
[image]

jonasmalacofilho, 2020-08-12:
My first guess would be that the device doesn't know what to do with `5a00...`.  Sometimes this can happen due to [missing padding](https://github.com/pyusb/pyusb/issues/126#issuecomment-179194578).

You may also be simply missing a suitable conversion, or do you really need to write a hexadecimal _string_ representation of some bytes?

```python
write(bytes.fromhex("5a00..."))
```

WanManito, 2020-08-12:
@jonasmalacofilho 
Converting to a byte array still gives the same error,In practice I'll use hexadecimal or an array of bytes

jonasmalacofilho, 2020-08-12:
Well, in that case, take a look if you don't need to pad the message in a certain way, or if you're writing to the wrong endpoint (devices can have multiple interfaces and endpoints, each using/supporting a different protocol).

If after that you still think there's a bug, please attach the complete output of this example code with the following environment variables set: `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug`.

---

P.S. "hexadecimal or an array of bytes"... Just be be clear, the device sometimes expects a hexadecimal string representation of some bytes, instead of the bytes themselves?  I suppose anything is possible, but that's very unusual (and inefficient).

WanManito, 2020-08-12:
[image]

WanManito, 2020-08-12:
The endpoints should be correct

WanManito, 2020-08-12:
DEVICE ID 03eb:2421 on Bus 001 Address 036 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x03eb
 idProduct              : 0x2421
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x1 ATMEL ASF
 iProduct               :    0x2 HID Generic, keyboard, CDC
 iSerialNumber          :    0x3 123123123123
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 488 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x29 (41 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xe0 Self Powered, Remote Wakeup
   bMaxPower            :   0xf4 (488 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x6 Error Accessing String
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2
      ENDPOINT 0x2: Interrupt OUT ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x2

WanManito, 2020-08-13:
@jonasmalacofilho The problem I found was that the 64 - and 65-byte lengths were not satisfied,Is there a way not to have to satisfy this byte length

jonasmalacofilho, 2020-08-13:
> Is there a way not to have to satisfy this byte length

It's really up to you to implement the protocol the device understands/expects (and devices don't always follow the spec).

closed 2020-08-13 by jonasmalacofilho