pyusb/pyusb #314: Operation timed out

WanManito, 2020-08-13:
# inPoint.read(64)
dev.read(0x81, 64, 0)

When I open a thread to read the data,Is there a way not to trigger this exception[Operation timed out],
Because the thread I'm opening is always getting the data
If a timeout is set, it will still be triggered if the data is not read when the time is up
To be precise, when you don't read the data,This exception is not raised

WanManito, 2020-08-13:
I also want to know how to release resources correctly，Because I don't have a method call like close,It caused me to fail to open another link. It should be occupied

jonasmalacofilho, 2020-08-13:
>  dev.read(0x81, 64, 0)

Please include the complete output with `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug`.

I would also like you to paste the complete code example (I'm assuming it's pretty similar to #313, but I would like to be sure).  Please paste it inline (with <code>```</code> markup) instead of using screenshots.

One more thing: what PyUSB version are you using?

---

> I also want to know how to release resources correctly，Because I don't have a method call like close,It caused me to fail to open another link. It should be occupied

Take a look at the [tutorial](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst#dont-be-selfish).

WanManito, 2020-08-13:
@jonasmalacofilho Sorry, I will not deal with the above problem for the moment.I didn't find a way in the tutorial to release the resource (close()),

WanManito, 2020-08-13:
pyusb==1.0.2
I'm using this version

WanManito, 2020-08-13:
@jonasmalacofilho 
usb.util.dispose_resources(dev)
I may have found a way to free up resources，
It's really useful to do it，
I don't know if that's the way it works，Or there are other ways

jonasmalacofilho, 2020-08-17:
Yes it is.  From the tutorial section linked above:

> If you want to claim and release interfaces manually, you may use the `claim_interface` and `release_interface` functions.  `claim_interface` will claim the specified interface if the device has not done it yet.  If the device already claimed the interface, it does nothing.  In a similar way, `release_interface` will release the specified interface if it is claimed.  If the interface is not claimed, it does nothing. You can use manual interface claim to solve the [configuration selection problem](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html#configsel) described in the [libusb](https://libusb.info/) documentation.
>
> If you want to free all resources allocated by the device object (including interfaces claimed), you can use the `dispose_resources` function.  It releases all resources allocated and puts the device object (but not the device hardware itself) in the state it was at the time when the `find` function returned.

closed 2020-08-17 by jonasmalacofilho

shuvragood, 2022-10-03:
ERROR: --- Failed to read. ---
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Avinash\OneDrive - hyderabad.bits-pilani.ac.in\Desktop\Number Of cycles\Potentiostat_SW_A9\Potentiostat_communication.py", line 80, in usb_read
    usb_input = self.ep_in.read(size, timeout)
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\core.py", line 301, in read
    return self.device.read(self.bEndpointAddress, size, self.interface, timeout)
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\core.py", line 656, in read
    return fn(
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 481, in bulk_read
    return self.__read(_lib.usb_bulk_read,
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 566, in __read
    ret = int(_check(fn(
  File "C:\Users\Avinash\AppData\Local\Programs\Python\Python310\lib\site-packages\usb\backend\libusb0.py", line 387, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n

jonasmalacofilho, 2022-10-03:
@shuvragood,

Please stop

- spamming our issue tracker, especially old/closed issues;
- posting tracebacks without accompanying text and context;
- ignoring our replies

at once.  This is your only warning.

If you want to report a bug _in PyUSB,_ please open a new issue.  On the other hand, if you just want to ask for some help, then use the discussions tab; but first consult our documentation (tutorial and FAQ), as well as try to find and read relevant past issues and discussions.  And in either case always _include context,_ especially information about your device, what are you trying to do, a code example, etc.