pyusb/pyusb #315: USBError: Resource busy

ravikuc, 2020-08-14:
I want to communicate with and send data to a USB device. I am able to find the device but while attaching the device with the kernel driver it is giving USB Error: Resource Busy. The following is my code:

`import usb

dev = usb.core.find(idVendor=0x22b8, idProduct=0x2e76)

reattach = False
if dev.is_kernel_driver_active(0):
    reattach = True
    dev.detach_kernel_driver(0)

dev.set_configuration()
cfg = dev.get_active_configuration()

interface_number = cfg[(0,0)].bInterfaceNumber
alternate_settting = usb.control.get_interface(dev, interface_number)
intf = usb.util.find_descriptor(cfg, bInterfaceNumber = interface_number,
                            bAlternateSetting = alternate_settting)

ep = usb.util.find_descriptor(intf,custom_match = \
      lambda e: \
    usb.util.endpoint_direction(e.bEndpointAddress) == \
    usb.util.ENDPOINT_OUT)
ep.write("test\n\r")

# This is needed to release interface, otherwise attach_kernel_driver fails
# due to "Resource busy"
usb.util.dispose_resources(dev)

# It may raise USBError if there's e.g. no kernel driver loaded at all
if reattach:
    dev.attach_kernel_driver(0)`

jonasmalacofilho, 2020-08-14:
What exactly is wrong with the (current) version of the code above?

ravikuc, 2020-08-14:
I am getting the below error when executed the above script:
usb.core.USBError: [Errno 16] Resource busy

jonasmalacofilho, 2020-08-14:
Hm, ok.  What confused me is that your first version of the code should a comment suggesting that `usb.util.dispose_resources(dev)` had already fixed the problem.

---

Regardless, let's take a look.  Please post the complete output with `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug`.

jonasmalacofilho, 2020-09-12:
Closing for now due to lack of feedback.

I'll be happy to take a look (and reopen the issue) once/if you post the output with `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug`.

closed 2020-09-12 by jonasmalacofilho

setaremalekiii, 2024-11-07:
I have the same issue above! However, it's specifically from the last line that I have commented out. I was just wondering why that is the case tho.. I am communicating with an STM32 h4 board

import sys
import usb.core
import usb.util
import time

devboard = usb.core.find(idVendor=0x0483, idProduct=0x5740)
interface = devboard[0][(1,0)]

endpointW = interface[0]
endpointR = interface[1]

while True:
    if devboard.is_kernel_driver_active(interface.bInterfaceNumber):
        try:
            devboard.detach_kernel_driver(interface.bInterfaceNumber)
            usb.util.claim_interface(devboard, interface.bInterfaceNumber)
        except usb.core.USBError as e:
            sys.exit("Could not detatch kernel driver from interface({0}): {1}".format(i, str(e)))

    dataIn = input()
    if dataIn == "exit":
        break
    bytesWritten = devboard.write(endpointW.bEndpointAddress, dataIn)
    print(f"Endpoint Address (Read): {endpointR.bEndpointAddress}")
    print(f"Endpoint Attributes: {endpointR.bmAttributes}")
    print(f"Endpoint Address (Read): {endpointW.bEndpointAddress}")
    print(f"Endpoint Attributes: {endpointW.bmAttributes}")

    print(f"Wrote {bytesWritten} bytes")
    time.sleep(1)
    buf = devboard.read(endpointR.bEndpointAddress, len(dataIn))

    res = bytes(buf).decode()
    print(res)
    #this is resulting in error number 2 so if you get err2 its cuz of the util stuff 
    
usb.util.release_interface(devboard, interface.bInterfaceNumber)
#I get resource busy errors when trying to reattach the kernel I think this will be an issue later on but right now its good? 
#devboard.attach_kernel_driver(interface.bInterfaceNumber)