pyusb/pyusb #316: USBError Accessing Configurations

jstiefel, 2020-08-17:
I'm trying to follow the "Let's get it started" tutorial for a lab power supply (Aim TTi CPX400SP). This works perfectly fine:

```
# find our device
dev = usb.core.find(idVendor=0xfffe, idProduct=0x0001)

# was it found?
if dev is None:
    raise ValueError('Device not found')
```
However, `dev.set_configuration()` fails with:

```
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 757, in get_configuration_descriptor
    config, byref(cfg)))
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
```
I'm executing it on Ubuntu 18 as sudo.

When I print out the instance object, I get this: 

```
sudo python3 cpx_test.py 
DEVICE ID 103e:0460 on Bus 003 Address 010 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x2 Communications Device
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x103e
 idProduct              : 0x0460
 bcdDevice              :  0x200 Device 2.0
 iManufacturer          :    0x4 TTi
 iProduct               :    0xc CPX Series PSU
 iSerialNumber          :   0x2a 434839
 bNumConfigurations     :    0x1 USBError Accessing Configurations
```
The last line is suspicious: `bNumConfigurations     :    0x1 USBError Accessing Configurations`. It seems like I can not access the configurations even as sudo. `cfg = dev.get_active_configuration()` fails with the exactly same error as mentioned above for `set_configuration()`. 

I also checked if the Kernel driver is active with `print(dev.is_kernel_driver_active(0))`, which outputs "False".

Is this access error caused by my USB device or some access restrictions in Ubuntu? Is there a way around this?

jonasmalacofilho, 2020-08-17:
At first glance it seemed to me that the device was, for some reason, refusing to respond to `GET_CONFIGURATION` / `SET_CONFIGURATION` requests.

However, in order to set the default configuration, `dev.set_configuration()` first fetches it with `libusb_get_config_descriptor`, and _that function is documented to **not** do any requests to the device._

<details><summary>call graph within PyUSB</summary>

```
Device.set_configuration(config: None)
+-- _ResourceManager.managed_set_configuration(config: None, ...)
    +-- Device.__getitem__(0)
    |   +-- Configuration(configuration: 0, ...)
    |       +-- IBackend.get_configuration_descriptor(config: 0, ...)
    |           +-- libusb_get_config_descriptor(config: 0, ...)
    |                   # Get a USB configuration descriptor based on its
    |                   # index. This is a non-blocking function which does not
    |                   # involve any requests being sent to the device.
    +-- _ResourceManager.managed_open
    |   +-- IBackend.open_device
    +-- IBackend.set_configuration
        +-- libusb_set_configuration
```
</details>

On the other hand, that function attempts to parse the configuration and [may return `LIBUSB_ERROR_IO`](https://github.com/libusb/libusb/blob/fa3f91e7c4c73fd233104cf3cad4890dd6121043/libusb/descriptor.c#L365) if it finds a problem there.

But since I haven't looked at all the code paths in libusb, the error could also come from another place.

Can you run two tests – one calling `dev.set_configuration()` and the other `dev.get_active_configuration()` – having first set `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4`?

---

Apart from you having mentioned using sudo, I can also see that transfers are reaching the device

```
 iSerialNumber          :   0x2a 434839    # requires/is the result of a GET_DESCRIPTOR request
```

so we can firmly rule out a permission error or a completely unresponsive device.

jstiefel, 2020-08-17:
@jonasmalacofilho Thank you for your quick response.

Here the outputs in Debug mode.

`dev.set_configuration()`:

```
sudo python3 cpx_test.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000001] [000018fe] libusb: debug [libusb_init] created default context
[ 0.000022] [000018fe] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.000043] [000018fe] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000059] [000018fe] libusb: debug [op_init] bulk continuation flag supported
[ 0.000065] [000018fe] libusb: debug [op_init] zero length packet flag supported
[ 0.000079] [000018fe] libusb: debug [op_init] sysfs can relate devices
[ 0.000084] [000018fe] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000240] [00001900] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.002617] [000018fe] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.002627] [000018fe] libusb: debug [linux_get_device_address] scan usb3
[ 0.002663] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.002670] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.002674] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.002744] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.002750] [000018fe] libusb: debug [linux_get_device_address] scan 3-2
[ 0.002770] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=7
[ 0.002774] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 7 session_id 775
[ 0.002778] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/7 (session 775)
[ 0.002797] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x26aea50 (3-2) has parent 0x26b33c0 (usb3) port 2
[ 0.002846] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6 detached: 0
[ 0.002850] [000018fe] libusb: debug [linux_get_device_address] scan 3-6
[ 0.002870] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.002873] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.002877] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.002895] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x2652b40 (3-6) has parent 0x26b33c0 (usb3) port 6
[ 0.002941] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6.1 detached: 0
[ 0.002945] [000018fe] libusb: debug [linux_get_device_address] scan 3-6.1
[ 0.002965] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.002968] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.002972] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.002990] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x2673f40 (3-6.1) has parent 0x2652b40 (3-6) port 1
[ 0.003035] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6.2 detached: 0
[ 0.003039] [000018fe] libusb: debug [linux_get_device_address] scan 3-6.2
[ 0.003061] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.003065] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.003069] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.003088] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x2676e80 (3-6.2) has parent 0x2652b40 (3-6) port 2
[ 0.003134] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6.4 detached: 0
[ 0.003139] [000018fe] libusb: debug [linux_get_device_address] scan 3-6.4
[ 0.003157] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=5
[ 0.003161] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 5 session_id 773
[ 0.003165] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/5 (session 773)
[ 0.003183] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x262c6b0 (3-6.4) has parent 0x2652b40 (3-6) port 4
[ 0.003224] [000018fe] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.003228] [000018fe] libusb: debug [linux_get_device_address] scan usb4
[ 0.003247] [000018fe] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.003251] [000018fe] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.003255] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.003338] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 4-5 detached: 0
[ 0.003346] [000018fe] libusb: debug [linux_get_device_address] scan 4-5
[ 0.003370] [000018fe] libusb: debug [linux_get_device_address] bus=4 dev=2
[ 0.003373] [000018fe] libusb: debug [linux_enumerate_device] busnum 4 devaddr 2 session_id 1026
[ 0.003378] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 4/2 (session 1026)
[ 0.003397] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x26a9780 (4-5) has parent 0x26a7070 (usb4) port 5
[ 0.003445] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 4-5.4 detached: 0
[... cut ...]
[ 0.083559] [00001926] libusb: debug [libusb_unref_device] destroy device 3.3
[ 0.083564] [00001926] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.083568] [00001926] libusb: debug [libusb_unref_device] destroy device 3.7
[ 0.083573] [00001926] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.083576] [00001926] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.083589] [00001926] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 0.083613] [00001928] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

jonasmalacofilho, 2020-08-17:
Thanks, this confirms that the problem is during parsing of the configuration descriptor.  Specifically (and if I understand it correctly), when the code reaches the endpoint the entire descriptor was supposed to already have ended.

I honestly don't know whether it's the device or libsusb that's at fault.  I gravitate towards blaming the former, but the earlier warning has to do with the size of the descriptor as well, and could be related.

I think we need some assistance from upstream here... do you mind reporting this to [libusb](https://github.com/libusb/libusb) directly?  Please ping me on the issue.

jonasmalacofilho, 2020-08-18:
@jstiefel,

Unfortunately the issue really lies in your device.

I would start by trying to get a firmware update from the manufacturer.  But you should know that it is _not_ very likely that it change/fix the USB descriptors.

The next step would be to see if there are (in kernel terminology) quirks that can be added so that these broken descriptors are accepted and reach libusb/PyUSB.  Finally, you could try to write your own kernel driver.

I'm closing this issue as well, for the same reasons mentioned in the libusb issue, but please feel free to ask more questions.

closed 2020-08-18 by jonasmalacofilho

jstiefel, 2020-08-18:
@jonasmalacofilho 

Thanks for your help. I even made a firmware update without any success. On the other side, I'm wondering why the firmware update over USB worked if USB is broken. Anyway, I'm going to give the LXI interface a try now.

jonasmalacofilho, 2020-08-18:
You  did the update on Windows, right?  Probably the device and update software were tested to (sort of) work on Windows, even though the device has broken descriptors.

jstiefel, 2020-08-19:
Exactly, I used the update tool for Windows.