pyusb/pyusb #317: Check dslr camera conection in windows

isdito, 2020-08-18:
Hello, I will like check the conection of one camera and take the path to the folder inside, but I dont know if with pyusb I could realease this. Could you help me

Give me a error
```
import usb.core
dev = usb.core.find()

File "usb-devices.py", line 2, in <module>
    dev = usb.core.find()
  File "C:\Python37\lib\site-packages\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

```

jonasmalacofilho, 2020-08-18:
You need to install libusb, and you may also need to install a compatible driver.  See:

 - https://github.com/pyusb/pyusb/blob/master/docs/faq.rst
 - https://github.com/libusb/libusb/wiki/Windows

isdito, 2020-08-18:
yes I find it.Thank you

 I'm looking for a way to get inside a USB-connected camera D750  this have a SD card, but it is not visible like a disk.

```

import usb.core
import usb.util

#pip install infi.devicemanager
#pip install libusb

from infi.devicemanager import DeviceManager
dm = DeviceManager()
devices = dm.all_devices
for i in devices:
    try:
        print ('{} : address: {}, bus: {}, location: {}'.format(i.friendly_name, i.address, i.bus_number, i.location))
    except Exception:
        pass

import usb.backend.libusb1

backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
dev = usb.core.find(backend=backend, find_all=True)

def EnumerateUSB():    #I use a simple function that scans all known USB connections and saves their info in the file
    with open("EnumerateUSBLog.txt", "w") as wf:
        counter = 0
        for d in dev:
            try:
                wf.write("USB Device number " + str(counter) + ":" + "\n")
                wf.write(d._get_full_descriptor_str() + "\n")
                wf.write(d.get_active_configuration() + "\n")
                wf.write("\n")
                counter += 1
            except NotImplementedError:
                wf.write("Device number " + str(counter) + "is busy." + "\n")
                wf.write("\n")
                counter += 1
            except usb.core.USBError:
                wf.write("Device number " + str(counter) + " is either disconnected or not found." + "\n")
                wf.write("\n")
                counter += 1
        wf.close()
```

This return 
```

Controlador de host eXtensible Intel(R) USB 3.0 - 1.0 (Microsoft) : address: 1310720, bus: 0, location: Bus PCI 0, dispositivo 20, función 0
Killer E2400 Gigabit Ethernet Controller : address: 0, bus: 4, location: Bus PCI 4, dispositivo 0, función 0
D750 : address: 9, bus: 0, location: Port_#0009.Hub_#0001
```

jonasmalacofilho, 2020-08-18:
I could  be wrong, but I don't think the SD card is visible through that interface like in a normal card reader.  As far as I remember that interface supports a lot of proprietary (or at least non standard) stuff for controlling the DSLR, accessing a virtual viewfinder, etc.

I have a D7000, I can try to hook it up and take a look.  But it will take me a while, I need to fish it out of storage and may (will probably) need new batteries, which could be a problem right now.

isdito, 2020-08-18:
Thank you @jonasmalacofilho yes this is the problem I not find normally the files. I only like search in the card and copy images to another folder... The rest of the options for me is the same. The problem it is windows in linux there is a very good library 

[image]

jonasmalacofilho, 2020-08-18:
If I understood your last comment correctly, you want to write a program that copies the images from the camera to some local folder (on Windows).

Assuming that's it, you don't need to deal with low level USB communication (which is the purpose of PyUSB) once you already have access to the images through the filesystem, as your screenshot shows.

isdito, 2020-08-19:
Good morning, yes and no. With pyusb I assume I know it's a camera that's connected, on the other hand if I need to get to that folder to get the images.

jonasmalacofilho, 2020-08-19:
You said that you need to figure out how Windows maps these specials folder to actual devices, but it isn't clear how this ties in with checking the USB descriptors; even when the device uses standard USB classes (or you know all the VID/PIDs you're looking for), this doesn't immediately tell you where Windows is going to place the pictures from that device in its directory hierarchy; that's a Windows thing, not an USB one.

Have you checked how the folder is (actually, not visually) named?  If so, what information do you need to find the correct folder (i.e. do you need the VID, the PID, the bus+address, the USB port?)?

isdito, 2020-08-19:
**_Have you checked how the folder is (actually, not visually) named?_** 

Good afternoon I don't know how to answer that question, I can't find anything on the internet that tells me how to get to those temporarily mounted folders. This Is that I need acces to this folders to give to the application.

**_If so, what information do you need to find the correct folder (i.e. do you need the VID, the PID, the bus+address, the USB port?)?_**

The truth is that I didn't think that the usb information would contain any information about where the files are going. I'm really going in blind.

Thank you for response to me

jonasmalacofilho, 2020-08-19:
> Good afternoon I don't know how to answer that question, I can't find anything on the internet that tells me how to get to those temporarily mounted folders. This Is that I need acces to this folders to give to the application.

Try to check the properties window for a picture in the camera, there's a location (or something like it) field that may show the real folder path.

Another thing you can do is drag the folder, or a picture from it, to a Command Prompt/PowerShell window... this should give you a path to that object.

isdito, 2020-08-19:
> Try to check the properties window for a picture in the camera, there's a location (or something like it) field that may show the real folder path.
>
> Another thing you can do is drag the folder, or a [...]

Hello, There is nothing in the properties and with a tkinter open a dialog I not see the folder.  I will check put a vídeo and see the folder path of the transfer... but not sure of that.

reggards

mcuee, 2020-12-24:
In any case, this is not a pyusb problem so close this for now.

closed 2020-12-24 by mcuee