pyusb/pyusb #320: Question Can pyusb be used to create device side code?

cromestant, 2020-08-23:
Thinking about what it would take to create a device (midi controller) running on a rasp-pi, that I can plug in via USB to another computer/midi controllable device via USB.
is pyusb a good place to start exploring this?

jonasmalacofilho, 2020-08-24:
Hi @cromestant,

Not really, that device would be a USB _gadget._

I've only looked into this for Arduinos, so I don't exactly how it works with the Raspberry Pi.  That said, Linux has a USB gadget API, but you'll probably also need to look into how to put your pi's USB controller in device mode (instead of the usual host mode).

Hope this helps a bit.

P.S. Let me know if you find a simple way to set this up on a pi, as it would be useful for #235.

closed 2020-08-24 by jonasmalacofilho

VCBE123, 2020-09-06:
I want to use my Jetson Xavier NX which runs a ubuntu system to simulate the gamepad signal and control the game on another windows 10 computer. So maybe I need to use the USB repo and send the message to the windows system. Is there any advice for me ?

mcuee, 2020-09-09:
For Linux, look at gadget driver.
https://www.kernel.org/doc/html/v4.17/driver-api/usb/gadget.html

mcuee, 2020-09-09:
https://raspberrypi.stackexchange.com/questions/71613/how-to-use-raspberry-pi-3-as-a-usb-gadget

Take note the following from the above answer:
You cannot use the Raspberry Pi 3 as a USB gadget. At the time of writing this is only possible on the Raspberry Pi 0, 0W, A, A+, and 4. The Pi B, B+, 2, 3, 3A+, and 3B+ lack the required pin for OTG.

mcuee, 2020-09-09:
@VCBE123 
The above link to an example on RASPBERRY PI ZERO. You can use it as a reference.
https://www.isticktoit.net/?p=1383

ReenigneArcher, 2020-09-28:
@mcuee anything similar for Windows? I am investigating a way to send gamepad signals to control a video game console from Windows 10. The code will all be in python 2.7.