pyusb/pyusb #322: OSError: exception: access violation writing 0x0000000000000024

WanManito, 2020-09-04:
[image]
[image]
hello，I don't know why I print this exception

mcuee, 2020-09-09:
Did you try the latest libusb Windows 1.0.23 release and latest pyusb?

WanManito, 2020-09-09:
@mcuee 
[image]
[image]
I think I should use the latest library

mcuee, 2020-09-09:
No you are not using the latest pyusb. You need to download latest git repo and install by yourself, not using pip.

mcuee, 2020-09-09:
Your issue may already got fixed in latest git.
https://github.com/pyusb/pyusb/issues/288

WanManito, 2020-09-09:
@mcuee 
[image]
I have solved the problem by installing Git, thank you

WanManito, 2020-09-09:
@mcuee Hello, I found a bug in the new version. In 1.0.2, I called“ usb.util.dispose_ resources( self.dev ）”Release the resources and connect again next time. There is no problem
When I update to the latest version of GIT, I still call "dispose" for the same code_ Resources ", which is then triggered in the child thread“ self.dev.read (self.__ Again, I reported an exception in 64“ self.dev.set_ Configuration() "is stuck and cannot be connected,,,
Again, I didn't change a single line of code, just changed the version of the library

WanManito, 2020-09-09:
The same code, just switching to the latest version, causes an exception

jonasmalacofilho, 2020-09-09:
> Again, I didn't change a single line of code, just changed the version of the library

Can you `git bisect` to a specific commit that causes this problem?

jonasmalacofilho, 2020-09-16:
Note that there's already a specific issue for the different behavior under git/1.1.0: #325.

jonasmalacofilho, 2020-10-25:
Problem has been solved, as mentioned in #325.

closed 2020-10-25 by jonasmalacofilho