pyusb/pyusb #324: Release 1.0.3

mcuee, 2020-09-09:
Maybe it is a good time to tag a new 1.0.3 release as there are enough bug fixes in the current git.

jonasmalacofilho, 2020-09-09:
I agree... I'll do this during the weekend.

Thanks @mcuee.

jonasmalacofilho, 2020-09-13:
:tada: 

Actually released as 1.1.0 due to 4134e911be88 ("Add USBTimeoutError").

closed 2020-09-13 by jonasmalacofilho

mcuee, 2020-09-14:
Great. Thanks.

jonasmalacofilho, 2020-09-14:
@mcuee , @SimplicityGuy , @tormodvolden

I don't have write access to the SourceForge project.  Can one of you please upload the [release sdist and associated signature](https://github.com/pyusb/pyusb/releases/tag/v1.1.0) there?

Besides these, the SourceForge git mirror is also outdated.  I think we should phase out that platform for everything but the mailling list (and, in the future, maybe even that), and redirect all users to this repository (or the website).

tormodvolden, 2020-09-14:
Sure, I am happy to do this. I have now uploaded the tarball and the signature, and updated the git mirror. It is surely some extra work to maintain a mirror but it is also nice to show that there are alternatives. Sourceforge also has the advantage that you can upload a proper release tarball and leave it at that - c.f. the confusing double tarballs at the github release link.

jonasmalacofilho, 2020-09-15:
Thanks!

I guess you're right, there are some advantages to keeping SF active.  And not being able to remove the auto-generated tarballs from GitHub is really annoying : \

That said, can you also edit the project and wiki pages to guide users towards this repository, the website or, when applicable, the [tutorial](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst) and [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst)?

mcuee, 2020-09-15:
@tormodvolden Can you add me and Jonas to the Sourceforge side? My Sourceforge user name is xiaofanc.

jonasmalacofilho, 2020-09-15:
Good idea, mine is `jonasmalaco`.

tormodvolden, 2020-09-15:
Done.